verilator/verilator #2305: make test fails if perl is not at /usr/bin/perl

timsnyder, 2020-05-04:
I'm trying to create a conda-forge recipe for verilator so that it can be installed using the conda package manager.

conda-forge tries to satisfy all dependencies in `$PREFIX`.  As such `/usr/bin/perl` doesn't exist on it's build host.

I will submit a PR that does the search and replace of `/usr/bin/perl` to `/usr/bin/env perl` (while also correctly handling `-w` in the shebang).

wsnyder, 2020-05-04:
Sorry, should have checked this earlier, but you missed a bunch, please grep again making sure to be using master.

timsnyder, 2020-05-04:
Ooh.  I should have checked that earlier too.  Will rerun my greps and oneliners again on master and open a new PR.

PR #2306 (2020-05-04): Use /usr/bin/env perl in lieu of /usr/bin/perl
Enables scripts to work where perl is not installed at /usr/bin/perl

Fixes #2305 

This PR didn't seem serious enough to warrant adding myself to `CONTRIBUTORS`
PR #2307 (2020-05-04): More /usr/bin/env perl in lieu of /usr/bin/perl
Enables scripts to work where perl is not installed at /usr/bin/perl

Previous commit was based on 4.028. This commit catches up to
tip of master.

Fixes #2305
closed 2020-05-04 by wsnyder