verilator/verilator #2309: DPI array ordering possibly wrong

teqdruid, 2020-05-05:
Hi Guys-

I've written a new test which passes in Questa but fails in Verilator. (Actually, it requires a few further unrelated modifications to work in Questa.)

The new test is in my fork/branch [here](https://github.com/teqdruid/verilator/tree/dev/teqdruid/byte_array_mismatch) and it's the t_dpi_open test.

So I'm trying to determine which of the three possibilities it is:
   1) My test is wrong -- possibly relying on DPI undefined behavior
   2) Questa has a bug
   3) Verilator has a bug

Can someone take a closer look?

wsnyder, 2020-05-05:
Passes in VCS.  @gezalore mind looking please?

gezalore, 2020-05-06:
The check that fails is realated to byte order in what the spec calls the  "actual reperesentattion":
https://github.com/teqdruid/verilator/blob/237133b166b2849844fe269987023770a51f9e74/test_regress/t/t_dpi_open_c.cpp#L274-L278
I think this is indeed implementation defined behaviour.

The relevant part of IEEE 1800-2018 is "H.12.4 Access to actual representation". This says that 
```
void *svGetArrayPtr(const svOpenArrayHandle);
int svSizeOfArray(const svOpenArrayHandle);
```
can return NULL/zero if accessing the internal representation of the array via a C pointer is not useful. In fact I believe it would be standard compliant if these always returned 0, they are just there to gain some efficiency when possible (i.e.: when the simulator internal representation happens to be something that is meaningully accessible via a C pointer). There is also a note in that section:

>NOTE—No specific representation of an array is assumed here; hence, all functions use a generic pointer void *. 

This means that the pointer you get is not necessarily useful without further implementation defined information about how to access it. In this particular case, the question is whether `byte x[0:N-1]` or `byte x[N-1:0]` maps to C `char x[N]` with identity index mapping. My suspicion (which I haven't verified) is that Verilator maps both of those SV array to the C array with identity index mapping, but Questa/VCS maps `byte x[0:N-1]` to `char x[N]` with identity indices but maps `byte x[N-1:0]` to with byte-reversed indices (hence your byte-reversed check passes on Questa). As this is implementation specified behaviour I don't think either is wrong. If it was up to me both `svGetArrayPtr` and `svSizeOfArray` would throw a run-time warning indicating they are implementatin defined.

If you want your code to be portable across simulator, you need to access array elements via the canonical representation using the acessors described in H.12.5, H.12.6 and H.12.7 (corresponding to varilous kinds of array element types). In this particular case, H.12.7 applies because `byte` is a C compatible type, in which case H.12.7 says that the element pointer accessors defined in H.12.4 *will* return a C compatible pointer. The portable solution hence I believe is:

```
// 'i' is the array handle defiend as 'const svOpenArrayHandle i'
for (int j=0; j<13; j++)
{
    CHECK_RESULT_HEX(*(char*)svGetArrElemPtr1(i, j), j);
}
```

I'm happy to discuss this further if you would like.

gezalore, 2020-05-06:
Note: It is possible there are still some non-compliant things in Verilator regarding open arrays. I indent to test these thoroughly further (hence #2302 for example), but that might take some time so please continue reporting any other things you think is suspect in the meantime.

teqdruid, 2020-05-06:
Well damn... We're relying on this implementation-specific behavior in some internal tools... Assuming "actual representation" in the spec means "simulator internal representation" you appear to be correct.

teqdruid, 2020-05-06:
I don't have access to NC but if it exhibits the same behavior as Questa/VCS, it may be better to change the layout which Verilator uses to be consistent with the major 3.

That might be a major change and I could see it breaking quite a few things, though.

closed 2020-05-06 by teqdruid