verilator/verilator #2311: Wrong comparison result with the != operator

srgatos6, 2020-05-06:
I wrote a serial multiplier and a test bench. The multiplier is correctly working and I compare the result against the product ( '*") operator as follows:

   always @(posedge done)
      begin
         tst = multiplicand * multiplier;
         if ( tst != result )
                tag = " error ***";
               else
                tag = "    ";
         $display(
                 "%d",$time,"         ",
                 "%d",multiplicand," ",
                 "%d",multiplier,"                 ",
                 "%d",result," ",
                 "%d",tst,"      ",
                 "%b",done," ",
                 "%s",tag," ",
                 "<-- result"
                );
        end

However when running a simulation even that the results are identical the tag is printed as "error ***" (see below). Even that the result 2 * -8 is equal to 16.

               40800           2 -8                  -16  -16      1  error *** <-- result

I worked around the problem by doing the comparison in a FORTRAN 2 style.
    always @(posedge done)
      begin
         tst = multiplicand * multiplier;
         if (!(tst-result))
           tag = " error ***";
         else
           tag = "    ";
         $display(
                 "%d",$time,"         ",
                 "%d",multiplicand," ",
                 "%d",multiplier,"                 ",
                 "%d",result," ",
                 "%d",tst,"      ",
                 "%b",done," ",
                 "%s",tag," ",
                 "<-- result"
                );
      end
And the problem does not show up.
               40800           2 -8                  -16  -16      1      <-- result

The problem also doesn't when using iverilator.

To reproduce the problem please compile using iverilog and verilog as follows:

make imult
make wmult
make mult

imult
wmult
mult

The third run (mult) will expose the problem.

The verilator version is

verilator --version
Verilator 4.016 2019-06-16 rev UNKNOWN_REV

This is what I get when install it by default in ubuntu as follows:

sudo apt-get update
sudo apt-get install verilator

Regards
S.R.

[serial_multiplier.tar.gz](https://github.com/verilator/verilator/files/4589552/serial_multiplier.tar.gz)

srgatos6, 2020-05-06:
Sorry I meant

The problem also doesn't when using **iverilog**

wsnyder, 2020-05-06:
```
`ifndef WORK_AROUND
         if ( tst != result )
`else
         if (!(tst-result))
`endif
```

How are these equivalent?  If tst = 1 and result = 0, then tst != result is true.  However  1-0 = 1 which notted is zero.

wsnyder, 2020-05-06:
BTW sorry you are having problems - assuming something else is wrong, please reduce the test case to the smallest possible and point at what to exactly see that is wrong, e.g. got this result from this expression and it is wrong.  Thanks.

srgatos6, 2020-05-06:
Yes you are right the work around is wrong.

Just compare the simulation between iverilog and verilator.

-1 * -7 == 7

Iverilog 
1017700 -1 -7 7 7 1 <-- result (no error reported)

verilator (mult no workaround defined
1017800 -1 -7 7 7 1 error *** <-- result it thinks that 7 != 7 and makes the tag error ***

Cheers

wsnyder, 2020-05-06:
I don't see any errors with mult on verilator, please try with git master.

srgatos6, 2020-05-06:
Here is a little test case that shows what I see. I'll try to get the git master tomorrow.

wsnyder, 2020-05-08:
Did you get a chance to try this on git master please?

srgatos6, 2020-05-09:
Not yet. I am writing UVM so I am using the EDA playground.
I'll try this weekend.

srgatos6, 2020-05-13:
Hi Wilson

My computer just got upgraded ot the latest version of Ubuntu and it upgraded verilator as follows:

verilator --version
Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1

I had on my todo list to install the top of trunk as per our discussion but it saved me some time.

The problem does not show up in that version (see below no error reported when printing the result).

Regards
Sergio

 1021800 1 0 0 0 -1 -3 0000 0000000000 0 3 3 1 
 1021900 1 0 0 0 -1 -3 0000 0000000000 0 3 3 1 
 1022000 1 0 0 1 -1 -2 0000 0000000000 0 3 3 1 
 1022100 1 0 1 0 -1 -2 0000 0000000000 0 3 3 1 
 1022200 1 0 2 0 -1 -2 1110 0000000000 0 3 3 0 
 1022300 1 0 3 0 -1 -2 0111 0000000000 0 3 3 0 
 1022400 1 0 4 0 -1 -2 0011 1111111110 -2 3 3 0 
 1022500 1 0 5 0 -1 -2 0001 1111111010 -6 3 3 0 
 1022600 1 0 6 0 -1 -2 0000 0000000010 2 3 3 0 
 1022700 1 0 0 0 -1 -2 0000 0000000010 2 3 3 0 
 1022700 -1 -2 2 2 1 <-- result

wsnyder, 2020-05-13:
Great, thanks for confirming.

closed 2020-05-13 by wsnyder

srgatos6, 2021-05-05:
Hi Wilson
I've been looking all over the documentation and cannot find an answer.
I am doing a comparison of verilator cycle simulator for my employer against commercial tools. Essentially we are learning on how multi-threading can speedup simulations.
I wrote e teat case. It is a highly parallel matrix multiplication (I can share when done ) and I've been testing it using VCS.
I am facing a problem compiling the test bench. All but one task compile. So here is my question.
In the test bench I write the multiplicand and multiplier matrices (I actually also write the result but it is just for testing accessibility) and then I start the matrix multiplication.
//Test bench 
initial
      begin
    bank = 0;
    select = 0;
    write  = 0;
        address_row = 0;
    address_col = 0;
    data = 0;
    start = 0;
        mux_control = 0;
    WAIT(2);    
    reset = 1;
    WAIT(2);
        reset = 0;
    WAIT(2);
    write = 0;
    WAIT(1);
    WRITE_REGISTERS(0,1);
    WAIT(1);
    WRITE_REGISTERS(1,101);
    WAIT(1);
    WRITE_REGISTERS(2,201);
    WAIT(1);
    WRITE_REGISTERS(3,301);
    WAIT(1);
    READ_REGISTERS(0);
    WAIT(1);
    READ_REGISTERS(1);
    WAIT(1);
    READ_REGISTERS(2);
    WAIT(1);
    READ_REGISTERS(3);
    WAIT(1);
    start = 1;
        mux_control = 1;
    WAIT(1);
        start = 0;
    WAIT(`SIZE*`SIZE);
    $finish;
     end // initial begin

The problem I am facing is with the WAIT task which is written

   task WAIT ;

      input n;
      integer n;

      integer i;

      begin
     for (i=0;i<n;i=i+1)
       begin
           @(posedge clock);
       end
      end

However verilator does not like the @(posedge clock) in an initial statement.
My question is:
Is there a way to advance the time in an initial statement like in the @(posedge clock) statement used in the WAIT task? Or do I have to re-write the test case as a state machine (difficult if I am multiplying a 1000x1000 matrix).
Thanks and regards
Sergio

 
 

Closed #2311.

wsnyder, 2021-05-05:
If you have a new problem please create a new issue, and please don't include the "...wrote" at the bottom as it looks confusing in github.  Anyhow, Verilator doesn't support waits in any form, yes, you need to rewrite it as a state machine (e.g. have a cycle counter and if (cycle==) in each section of your wait statement.

srgatos6, 2021-05-06:
Got ya. However can that if cycle == be put in an "initial" statement. I'd like to keep the structure of the test in the initial statement I showed you on my previous email.
As per re-using the previous email thread. I'll create a new issue in the future.
ThanksSergio