verilator/verilator #2314: Support dynamic arrays of unpacked arrays

gezalore, 2020-05-08:
The standard allows definitions like `bit dyn [][1:0]`, which is a dynamic array of upacked arrays (see examples In section 7.5.1 of IEEE 1800-2017). These are currently unspoorted but work on other tools. Queues of unpacked arrays are somewhat less interesting as I don't think there is a way of adding an element to them using the standard queue API.

wsnyder, 2020-12-13:
Today added array of queues. Added t_dynarray_unpacked.v to try this and the base declarations now work, but array subscripts are broken and looks like several places assumed arrays are always only of a basic type, or an array of a basic type.

wsnyder, 2023-08-30:
Fixed a while ago, just enabled test.

closed 2023-08-30 by wsnyder