verilator/verilator #2315: Support passing dynamic arrays and queues to DPI open array arguments

gezalore, 2020-05-08:
The current implementation of DPI open arrays does not support passing dynamic arrays and queues (this causes the compile time error #2293). The  t_dpi_open_query test has some examples of what is valid, but these are currently disabled for Verilator.

wsnyder, 2020-12-13:
@yTakatsukasa you've been in DPI recently, so no pressure but just FYI if interested.

Now that dynarrays/queues are generally in the code basics exist. Complication is array sizes will need to be dynamically instead of statically be computed in DPI wrapper, then all the dpi calls need to know it's a queue and call appropriate structures instead of arrays.  Note queues and dynamic arrays use same base data types so should be a two-for-one.

wsnyder, 2020-12-13:
Sleeping queues.on this, if we add queues/dynarrays we'll also perhaps want to handle mixes a e.g. queue-packed-dynarray mixes which means all of the access code will be much slower with pointer dereferences rather than math.  Probably better to avoid supporting queue/dynarrays until there's a strong need.

closed 2020-12-13 by wsnyder