verilator/verilator #2317: Include VM_GLOBAL to static __ALL.a

lromor, 2020-05-08:
Thanks for taking the time to report this.

What would you like added/supported?

I'm not sure if there's a reason of why the static library generated does not include also the other objects such as verilated.o or verilated-dpi.o. My feature request would be to add a flag to allow this behavior. This would simplify the linking to external libraries.

May we assist you in trying to fix this yourself?

I would be happy to provide a pull request if the feature makes sense.

gezalore, 2020-05-08:
I think the idea might have been that you can verilate multiple models and then link them all into a single executable  by linking against the multiple __ALL.a libraries, in which case you still need only one copy of the support objects, but I am no authority about this.

As `verilated.mk` uses `+=` with `VK_OJBS` (which holds the object that go into the __ALL.a), you can probaly add `VK_OBJS = $(VK_GLOBAL_OBJS)` to your Makefile to get that behaviour. As of whether this should be a separate option, I will defer to @wsnyder.

lromor, 2020-05-08:
> I think the idea might have been that you can verilate multiple models and then link them all into a single executable by linking against the multiple __ALL.a libraries, in which case you still need only one copy of the support objects, but I am no authority about this.

That's also something I'm looking for. Right now the documentation seems to discourage multiple top modules. Nonetheless I would like to test a single top module at a time, compile them all together in a single __ALL.a. Is this already supported?

lromor, 2020-05-08:
Thank you for your answer. Is it even possible without including one of the .mk inside my Makefile, to access to the generated set of $(VK_GLOBAL_OBJS)?

Right now I am manually listing the dependencies, for instance:
```
make -C obj_dir -f VModuleName.mk verilated.o
```

lromor, 2020-05-08:
> I think the idea might have been that you can verilate multiple models and then link them all into a single executable by linking against the multiple __ALL.a libraries, in which case you still need only one copy of the support objects, but I am no authority about this.

Would it be a bad idea to provide a shared and static library containing those support objects? In this way, it will be compiled only once and shared amongst different linkages.

gezalore, 2020-05-08:
This should work:
```
$ cat foo.mk
VK_OBJS += $(VK_GLOBAL_OBJS)
$ make -f VModuleName.mk -f foo.mk VModuleName__ALL.a
```

You can also externally merge tthe libs with something like https://stackoverflow.com/questions/3821916/how-to-merge-two-ar-static-libraries-into-one

The compilation of the support objects depend on flags used when you invoked verilator (via the generated makefiles), so they need to be compiled with the model. Also for the same reason compiling multiple top-levels with different Verilator flags to be used in the same exe is not recommended.

lromor, 2020-05-08:
Nice! Thank you!

wsnyder, 2020-05-08:
I'd note the most common usage is the --exe flag in which case a binary is made and no .a is exposed to the user.
As Giza noted, the "large system" usage is to have many .a's from multiple Verilator runs, then link with one support library.
For other usages, making your own makefile is the current suggested route.
If we think many people will use it we can add a flag, but not sure how common it would be used, and all flags add some complication if only to explaining them.

Are you OK with the proposed solution?

lromor, 2020-05-09:
Yes, I understand, and thank you for the clarification. I am OK with the proposed solution.

closed 2020-05-09 by lromor