verilator/verilator #2324: Support display/scan %u/%x

wsnyder, 2020-05-13:
We should implement display/scan %u and %x.

These [sv-tests](https://github.com/SymbiFlow/sv-tests) fail as %u/%x are not supported.

* generated/ivtest/scanu__fscanf_u.v
* generated/ivtest/scanu__sscanf_u.v
* generated/ivtest/scanz__fscanf_z.v
* generated/ivtest/scanz__sscanf_z.v

Todo:

1. Add test_regress tests for %u/%x and make sure expected values match what another simulator e.g. Icarus outputs.

2. Update src and include files to support these.  As Verilator runtime doesn't support tristates, the tristate part of %x would just be zero/ignored.

wsnyder, 2020-05-13:
Attn @stephenry this is one close to your current & appreciated efforts.

stephenry, 2020-05-14:
I'll take a look at it. Thanks.

wsnyder, 2020-05-18:
Thanks for your work on this.

closed 2020-05-18 by wsnyder