verilator/verilator #2327: Support unbounded as array index

wsnyder, 2020-05-13:
Good medium-starter task

#2326 will add support for queue slices which in some cases use unbounded.  It is also legal to use $ to refer to the maximum index of an array, e.g.

   reg [3:0] foo;
   reg [3:1] bar_is_foo_3_to_1 = foo[$:1];

Add appropriate packed and unpacked and multidimensional cases that use $ and update (probably) V3WidthSel and V3Width to support these - basically the Unbounded would be seen and the existing array bound which is then known would replace the $.

ShraddhaDevaiya, 2020-09-14:
Hi @wsnyder , how can I compile and simulate standalone SystemVerilog file ? For this example, I have tried with this command, ``` verilator --cc simple_ex1.sv```  and the code in the simple_ex1.sv file is following : 
```
module test;
reg [3:0] foo;
//reg [3:1] bar_is_foo_3_to_1 = foo[$:1];
initial begin
$display("WORKING !");
end
endmodule
```
Compilation is working, but it is not showing the display output. Am I missing something in command to simulate, Can you please help?

wsnyder, 2020-09-14:
Please it is more polite to make new issues instead of posting to old ones. But to answer you need to use --exe. make and also run the design. Please see the examples in the documentation, or the examples/ directories.

ShraddhaDevaiya, 2020-09-15:
Thanks, will make sure it.

wsnyder, 2020-11-07:
The syntax suggested here is not IEEE-compliant so not needed, unbounded can only be used with queues and assertions.

closed 2020-11-07 by wsnyder