verilator/verilator #2328: Support pattern member assignment with data type

wsnyder, 2020-05-14:
Medium starter task that could be straight-forward or could get complicated - not sure ;)

This [sv-tests](https://github.com/SymbiFlow/sv-tests) fail:

* tests/chapter-5/5.10-structures.sv 

Because Verilator should support pattern members with data types, e.g.

    ms = '{ int:0, int:1};

To fix this, 1. add appropriate tests. 2. Comment in the verilog.y has a patternKey rule for simple_type  to make an appropriate Ast type. 3. the pattern matching rules need to look for appropriate types.

Aj0SK, 2021-06-21:
Hi, during my work on this issue I found two things that are not clear to me.

1. I would consider adding some warning in case of multiple defaults, like here:
```
typedef struct packed { // [3:0]
      bit       b3;
      bit       b2;
      bit       b1;
      bit       b0;
   } b4_t;
const b4_t w = '{default: 1'b1, default: 1'b1};
```
From my understanding of code, I expected that this is caught [here](https://github.com/verilator/verilator/blob/ec1c1127914bd3df4e9a7441051e920af0fef182/src/V3Width.cpp#L3303) but the unlinks somehow skips the defaults.
2. Should this feature be allowed only for simple_type? Not also for other types?

wsnyder, 2021-06-27:
1. You are probably right, looks like there is no test yet for this error of multiple defaults so perhaps can you please add it and debug?
2. IEEE says this is legal only on simple types.

With this, the grammer fixed, and tests passing hopefully could merge!

closed 2022-07-25 by wsnyder