verilator/verilator #2329: Support $writememb

wsnyder, 2020-05-14:
Currently only $writememh (hex) is supported, we should also support $writememb (binary) for compliance.

Need to clone the current hex format test(s), update src/ methods, and verilated.cpp implementation.

FanShupei, 2020-06-26:
I'd like to work on this. Since I'm new to verilator codebase, it may take me some time to finish it.

wsnyder, 2020-06-28:
@FanShupei thanks for looking into this - no rush.  Please feel free to ask questions, it should be fairly straight forward since readmemb already works so you can follow a similar pattern.  Please also make sure to update the tests also to check these files, e.g. by adding test cases similar to writememh.

wsnyder, 2020-07-01:
Thanks for your contributions on this - hope you'll consider more.

PR #2450 (2020-07-01): Add $writememb support
Resolves #2329 

The implementation is quite straightfowrad. Unlike hex number, binary number cannot be formatted by printf directly, so I write a helper function `formatBinary` to do this.

For tests, it seems $writememh is tested in `t_sys_writemem`. I duplicate it to `t_sys_writemem_b` and add an option `WRITEMEM_BIN` to control whether $writememh or $writememb is used. Test `t_sys_readmem` is also modified to support the new added option. The corresponding golden output for binary case is also added.

I have tested `t_sys_writemem` and `t_sys_writemem_b` locally. It  [... cut]
closed 2020-07-01 by wsnyder