verilator/verilator #2333: 1a0da2e4 violates clang thread safety checks

gezalore, 2020-05-15:
@stephenry would you mind taking a look please?

You need to configure with clang and promoted warnings:
```
 ./configure --enable-longtests --enable-ccwarn CXX=clang++
```
Then try:
```
cd test_regress
./t/t_alw_split.pl --vltmt
```

Should say:
```
In file included from /home/gl/work/verilator/src/test_regress/../include/verilated.cpp:25:
/home/gl/work/verilator/src/test_regress/../include/verilated_imp.h:534:41: error: reading variable 'm_fdps' requires holding mutex 's_s.m_fdMutex' [-Werror,-Wthread-safety-analysis]
                if (VL_LIKELY(idx < s_s.m_fdps.size()))
                                        ^
/home/gl/work/verilator/src/test_regress/../include/verilated_imp.h:535:37: error: reading variable 'm_fdps' requires holding mutex 's_s.m_fdMutex' [-Werror,-Wthread-safety-analysis]
                    fp[out++] = s_s.m_fdps[idx];
                                    ^
/home/gl/work/verilator/src/test_regress/../include/verilated_imp.h:541:57: error: reading variable 'm_fdps' requires holding mutex 's_s.m_fdMutex' [-Werror,-Wthread-safety-analysis]
                if (fdi & VL_MASK_I(1)) fp[out++] = s_s.m_fdps[i];
                                                        ^
```

stephenry, 2020-05-15:
My apologies. I'm running a regression now, but It should be fixed in: https://github.com/stephenry/verilator/commit/23d7b857952512bb5e48c4a73501f649d8480564. 

Executing: `test_regress/driver.pl -vltmt` invokes the multi-threaded regression suite. Is that correct?

gezalore, 2020-05-15:
Thanks, I will commit that shortly.

> Executing: test_regress/driver.pl -vltmt invokes the multi-threaded regression suite. Is that correct?

That's right, thoguh I just run `make test_regress` which runs botht he MT and regular suite.

stephenry, 2020-05-15:
Actually, I may need to re-think this. In the MCD case, a write is essentially broadcast across multiple file-descriptors, and these file descriptors may themselves be owned by different threads. A thread could potentially close a file-descriptor in the small interview when this write broadcast takes place. This is unique to MCD file descriptors. It may be necessary to be overly pessimistic and place a lock over any write to a MCD descriptor. Let me review the implications.

gezalore, 2020-05-15:
Ok, take your time. Please PR the fix when you think it's ready.

wsnyder, 2020-05-19:
Note I added a LockGuard to fdToFp just so regressions would be able to pass for now.  I'm upgrading the clang runs to nightly so we'll catch faster.  This also had to move a couple of lock locations which might or might not be right when you fix the meta issue.

wsnyder, 2020-05-24:
@stephenry might it be possible to get this repaired by June 1?  I'd like it cleaned up in time for the next release if possible.  Thanks.

stephenry, 2020-05-24:
Sure. Give me a couple of days and I'll get back to you.

wsnyder, 2020-06-03:
Resolved with earlier merge.

closed 2020-06-03 by wsnyder