verilator/verilator #2334: compile verilator with uvm-systemc

ruiqicao, 2020-05-15:
Hi:
I would like to know is there any way to compile verilator with uvm-systemc. I just writed a basic sv file and utilize uvm-systemc file as testbench. When I compile them with usual command, like "verilator -Wall --cc --trace top.sv --exe top_tb.cpp make -j -C obj_dir/ -f Vtop.mk Vtop", then there is a bug saying there is no Vtop,mk.
Therefore, I guess maybe the compile command has something wrong or I compiled at a wrong way.

wsnyder, 2020-05-15:
Perhaps your module name isn't "module top" in top.sv?

ruiqicao, 2020-05-16:
Actually it is. And the program passed the first command "verilator -Wall --cc --trace top.sv --exe top_tb.cpp"

wsnyder, 2020-05-16:
Is there a .mk in obj_dir?

If it's not obvious what name is being used, use '--prefix Vtop' and/or attach the output of --debug.

ruiqicao, 2020-05-16:
Oh, I find the .mk in my obj_dir. I guess the problem is mainly the naming issues. Thank you so much.

wsnyder, 2020-05-16:
Great

closed 2020-05-16 by wsnyder

ruiqicao, 2020-05-16:
But for debugging, do you mean that I change the command to this: make -j -C obj_dir/ -f --prefix Vtop --debug. Maybe I could use it next time.

wsnyder, 2020-05-16:
No you pass "--prefix Vtop --debug" to verilator.

ruiqicao, 2020-05-16:
Got it. Thank you!