verilator/verilator #2335: OS X/clang C++ flags in the generated makefile

horvathp, 2020-05-16:
Hello, 

I'm using Verilator on OS X. I'm not sure if the issue described here is intentional or more of a user error.
Verilator version
Verilator 4.035
compiled with
Apple clang version 11.0.3 (clang-1103.0.32.59)
Verilator configures and builds itself with
`-std=gnu++14`
however, this option does not propagate to the generated Makefiles. This breaks e.g. --trace runs as shown below. Of course passing the proper CFLAGS to verilator, or adding them manually to the Makefile fixes the error. Would it be more appropriate to bake in the same -std value into the generated makefiles, given that stuff will get compiled together at source level?

```
In file included from /usr/local/share/verilator/include/verilated_vcd_c.h:24:
In file included from /usr/local/share/verilator/include/verilated_trace.h:27:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/string:504:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/string_view:175:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__string:56:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/algorithm:643:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:656:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/new:163:6: error: enumeration previously
      declared as scoped
enum align_val_t { __zero = 0, __max = (size_t)-1 };
     ^
note: previous declaration is here
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/new:203:102: error: no type named
      'align_val_t' in namespace 'std'
  ..._LIBCPP_OVERRIDABLE_FUNC_VIS void* operator new(std::size_t __sz, std::align_val_t) _THROW_BAD_A...
                                                                       ~~~~~^
```

gezalore, 2020-05-16:
Unles you use threads the generted code and the run-time library (in verilator/include) should not depend on C++11 hence the configured language version option is not passed through unless using threads. You can force this with VM_C11=1 to `make`. We do pass `-faligned-new` though so I wonder if your versino of libc++ gets tripped up by the combination as it seems like a plain redefinition error. Sorry I am not more more helpful but I don't have access to OS X. If you can investigate further it would be very helpful.

@wsnyder although it's not quite September yet, but should we just turn C++11 always on for OS X? I saw you had this come up already in #2298.

wsnyder, 2020-05-16:
I would be ok with always using C++11 with OS-X.

BTW still looking for someone with OS X knowledge to get OS X into regressions which would probably help make this more stable - #1477.

horvathp, 2020-05-16:
I have not tried forcing c++11. To be clear, I did not force C++14. When building Verilator itself, configure checks if std=gnu++14 works, then moves on and keeps -std=gnu++14 among the CFLAGS:
```
...
checking for inline... inline
checking whether g++ accepts -std=gnu++14... yes
checking whether g++ accepts -std=c++03... yes
checking whether g++ accepts -Wextra... yes
...
```
As far as I see, align_val_t is C++17, so I'm not sure why it even shows up. This might be some internal weirdness around Apple's C++ libraries? I wanted to try C++11 but I cannot force C++11 by adding -std=c++11 to CXXFLAGS before running configure; it sticks happily to C++14 as shown above.

Library dependencies of the compiled binary look just fine to me:
```
hp@ obj_dir % otool -L Vhelloworld
Vhelloworld:
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

gezalore, 2020-05-16:
The verilator compiler (i.e.: bin/verilator_bin) is always build againt whatever configure discovers. The verailated model (i.e.: the generated code output by bin/verilator + it's run-time dependencies which are shipped in /usr/local/share/verilator/include in your case) are build without C++11 by default. As I mentioned align_val_t is likely due to `-faligned-new` which we use if available at configure time, even with older standards. You can force C++11 for the verilated model by building the output of verilator with something like `make -C obj_dir Vhelloworld.mk VM_C11=1`.

horvathp, 2020-05-16:
Oh, OK, it makes sense, I didn't pay attention to the `-faligned-new` part. Adding `VM_C11=1` avoids the issue as you suggest. (Or passing -std=c++14 to verilator before, as I did).

Just out of curiosity, I tried to disable checking for a C++14-conform compiler in configure.ac. Sure enough, configure adds `-std=gnu++11` but there are literally dozens if not hundreds of instances in various headers where string is used as opposed to std::string:
`../V3Options.h:310:5: error: unknown type name 'string'; did you mean 'std::string'?`
and clang being picky as it is, hangs on them. Does it make sense to "fix" these?

wsnyder, 2020-05-16:
config_build.h has:
```
typedef std::string string;
using std::cout;
using std::endl;
using std::make_pair;
```

So while perhaps not the friendliest, with that I don't understand why the compiler would object to using "string" under any version of C++.

horvathp, 2020-05-16:
Indeed, I might have messed up config_build.h while tinkering. It builds cleanly with -std=c++11.
Thank you.

horvathp, 2020-05-17:
@wsnyder I had some success getting Travis to work on OS X. The build passes (it was pretty easy), but the tests fail. 
https://travis-ci.com/github/horvathp/verilator/builds/166572829
The bare minimum to be able to build is to install flex and bison (probably flex would be enough) 
as the older Apple-supplied flex (or bison?) liberally sprinkles "register" keywords, which will break the build because of -Werror.
 
My whole experience with Travis spans the last 15, so I'm not sure I did things the most practical and elegant way. The bigger problem is that I don't really understand the tests themselves, but I hope this can be a useful start nevertheless. 
Most tests fail like this (and this does not show up when I run it locally):
`
In file included from <built-in>:377:
<command line>:4:9: error: 'VM_COVERAGE' macro redefined [-Werror,-Wmacro-redefined]
#define VM_COVERAGE 0
        ^
<command line>:1:9: note: previous definition is here
#define VM_COVERAGE 
        ^
<command line>:5:9: error: 'VM_SC' macro redefined [-Werror,-Wmacro-redefined]
#define VM_SC 0
        ^
<command line>:2:9: note: previous definition is here
#define VM_SC 
        ^
<command line>:6:9: error: 'VM_TRACE' macro redefined [-Werror,-Wmacro-redefined]
#define VM_TRACE 0
        ^
<command line>:3:9: note: previous definition is here
#define VM_TRACE 
        ^
`

wsnyder, 2020-05-17:
I think it's saying VM_COVERAGE etc is on the command line twice...

     g++  -I/usr/local/opt/flex/include -I. -Wall -Wextra -Wfloat-conversion -Wthread-safety -Werror -MMD -I/Users/travis/build/horvathp/verilator/test_regress/../include -I/Users/travis/build/horvathp/verilator/test_regress/../include/vltstd -DVM_COVERAGE= -DVM_SC= -DVM_TRACE= -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -I. -Wall -Wextra -Wfloat-conversion -Wthread-safety -Werror -MMD -I/Users/travis/build/horvathp/verilator/test_regress/../include -I/Users/travis/build/horvathp/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o Vtop__ALLslow.o Vtop__ALLslow.cpp

It is, and with conflicting settings hence the error.  Something must be different in the configure or make, perhaps make a fake run to print out all of the makefile flags to see where it is coming from .

horvathp, 2020-05-17:
Yes, it looks that the entire argument list gets added twice, first with empty -DVM_COVERAGE values, then with meaningful ones. It does not happen locally. Could you maybe briefly summarize how this argument list gets handled during the tests? I'm a bit lost how the makefile-perl-cmake-verilator machinery pass around these things? What would be appropriate places to "instrument" during dry runs?

wsnyder, 2020-05-17:
e.g.

```
diff --git a/include/verilated.mk.in b/include/verilated.mk.in
index bd26a59c8..46ceaeb16 100644
--- a/include/verilated.mk.in
+++ b/include/verilated.mk.in
@@ -203,16 +203,20 @@ $(VM_PREFIX)__ALL.a: $(VK_OBJS)

 ifneq ($(VM_DEFAULT_RULES),0)
 $(VM_PREFIX)__ALLfast.o: $(VM_PREFIX)__ALLfast.cpp
+       $(MAKE) debug-make
        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_FAST) -c -o $@ $<

 $(VM_PREFIX)__ALLslow.o: $(VM_PREFIX)__ALLslow.cpp
+       $(MAKE) debug-make
        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_SLOW) -c -o $@ $<

 # VM_GLOBAL_FAST files including verilated.o use this rule
 %.o: %.cpp
+       $(MAKE) debug-make
        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_FAST) -c -o $@ $<

 %__Slow.o: %__Slow.cpp
+       $(MAKE) debug-make
        $(OBJCACHE) $(CXX) $(CXXFLAGS) $(CPPFLAGS) $(OPT_SLOW) -c -o $@ $<

 end
```if

gezalore, 2020-06-24:
Travis cron jobs now runs an OS X build and seems like your original issue is resolved. Closing this but feel free to reopen if issues remain.