verilator/verilator #2337: Output format flag

evgeniyBolnov, 2020-05-16:
```
4.030
%Error: delay.sv:18: syntax error, unexpected else
    else
    ^~~~
%Error: Exiting due to 1 error(s)

4.032
%Error: delay.sv:18:5: syntax error, unexpected else
   18 |     else
      |     ^~~~
%Error: Exiting due to 1 error(s)
```
please add a flag that allows you to change the output format.

wsnyder, 2020-05-16:
Sorry this was to match GCC's changing.

Need more details of what's breaking with the new format, and what you are looking for.

evgeniyBolnov, 2020-05-17:
There are many plugins for integration into various programs (Sublime, VSCode and others). Accordingly, they processed the verilator output and redirected it to the program. As you can see in the initial message in version 32, the line number and "|". As you can see in the initial message in version 32, the line number and "|" were added. In this regard, the question is: is it possible to add a flag for backward compatibility?

wsnyder, 2020-05-17:
Does -Wno-context fix it?  Most tools don't want the context info at all.

If not will look into having a format option.

wsnyder, 2020-05-26:
?