verilator/verilator #2339: 1a0da2e4ec90f338e8fc888d0357db78ec74f64a Introduces C++11 dependency

gezalore, 2020-05-17:
Apart from #2333, 1a0da2e4ec90f338e8fc888d0357db78ec74f64a also introduces a hard dependency of the model on C++11 via at least:
https://github.com/verilator/verilator/blob/212aa332dd9f14d93b897d41e56e4c28389ac117/include/verilated_imp.h#L245
https://github.com/verilator/verilator/blob/212aa332dd9f14d93b897d41e56e4c28389ac117/include/verilated_imp.h#L475

gezalore, 2020-05-17:
@stephenry would you mind looking at this too please? Currently Verilator still needs to support pre C++11 compilers whenever possible, both in verialtor itself and the verilated model.

stephenry, 2020-05-17:
This branch resolves the regression: https://github.com/stephenry/verilator/commit/b15b19cbff7281197244e128a836b0a0f475b362.

My environment defaults to C++11 which is why the issue was missed. I updated CPPCHECK, but this doesn't appear to catch the issue either. If I forcibly run a build under C++03, I note that there is another function `va_copy` which causes the build to fail. According to https://en.cppreference.com/w/cpp/utility/variadic/va_copy, this is C++11.

https://github.com/verilator/verilator/blob/dc25e9b949c7dcaf7f08d21191f060e4c98ebd31/include/verilated.cpp#L170-L175

I am delighted to report that I am not responsible for `va_copy` issue.

wsnyder, 2020-05-17:
Fortunately va_copy is in gcc 4.4.7/clib which is the oldest we need to go.  Can you please turn the patch into a pull request - thanks.

closed 2020-05-17 by wsnyder