verilator/verilator #2341: compile uvm-systemc

ruiqicao, 2020-05-17:
Still, I have some questions about compiling uvm-systemc. When I pass the command"make -j -C obj_dir/ -f Vtop.mk Vtop", Verilator reported error, such as " *** Unsupported: GNU Make cannot build in directories containing spaces". I checked the verilator include file, but I don't know the reason about this error.
Could anybody give some hints?

wsnyder, 2020-05-18:
Your current directory (e.g. what 'pwd' prints) or the directory you told Verilator to use presumably contains spaces, don't do that.

closed 2020-05-18 by wsnyder

ruiqicao, 2020-05-18:
Got it. And I would like to know if there is an uvm library. Because when I include <uvm>, it shows that no such file or directory.

ruiqicao, 2020-05-18:
like doing this: #include <uvm>

wsnyder, 2020-05-18:
Verilator does not come/is not related to SystemC UVM, sorry but I do not use it, and don't know how to install it, but suspect you can search online for assistance.

ruiqicao, 2020-05-18:
I understand. Thank you.