verilator/verilator #2344: VPI parameter & localparam

elgorwi, 2020-05-20:
I was wondering if parameters and/or localparams can be accessed through the VPI interface?
Had a quick look at ```verilated_vpi.cpp```, but couldn't tell immediately if this has been implemented or not?

Thanks

wsnyder, 2020-05-20:
Quick experiment suggests not.  If this matters to you please rename this bug to e.g. "Support VPI parameters & localparam". If you can/are willing we'd appreciate a pull request to fix this.

```
diff --git a/test_regress/t/t_vpi_var.v b/test_regress/t/t_vpi_var.v
--- a/test_regress/t/t_vpi_var.v
+++ b/test_regress/t/t_vpi_var.v
@@ -108,7 +108,7 @@ endmodule : sub

 module arr;

-   parameter LENGTH = 1;
+   parameter LENGTH /*verilator public_flat_rw*/ = 1;

    reg [LENGTH-1:0] sig /*verilator public_flat_rw*/;
    reg [LENGTH-1:0] rfr /*verilator public_flat_rw*/;
diff --git a/test_regress/t/t_vpi_var.cpp b/test_regress/t/t_vpi_var.cpp
--- a/test_regress/t/t_vpi_var.cpp
+++ b/test_regress/t/t_vpi_var.cpp
@@ -535,6 +535,7 @@ int _mon_check_putget_str(p_cb_data cb_data) {
                             = vpi_handle_by_name((PLI_BYTE8*)"check", data[i].scope));
             CHECK_RESULT_NZ(data[i].verbose
                             = vpi_handle_by_name((PLI_BYTE8*)"verbose", data[i].scope));
+            CHECK_RESULT_NZ(vpi_handle_by_name((PLI_BYTE8*)"LENGTH", data[i].scope));
         }

         static t_cb_data cb_data;

```

elgorwi, 2020-05-21:
> Quick experiment suggests not. If this matters to you please rename this bug to e.g. "Support VPI parameters & localparam". If you can/are willing we'd appreciate a pull request to fix this.
>
> ``` [...]

Thanks for confirming. The end goal would be making parameters & localparams accessible from cocotb. This feature would bring verilator more in line with the cocotb support of Icarus.

I'm not familiar with the verilator codebase, so if you could make some suggestions on how to approach adding the VPI support, I'll free up some time to work on a patch.

Thanks

wsnyder, 2020-05-22:
See docs/internals.adoc for general tips.

Process:
1. Add test to test_regress, the above would work (though need both parameter & localparam), but you might want just for testing a very small test case with just a parameter and a localparam.
2. Run verilator --debug and see how far the data gets, might need to suppress current removal of parameters.
3. In V3EmitC.cpp make sure is getting dumped into VPI structures in C code.
4. In verilated_imp.h check that parameters are represented in structures (that V3EmitC will make).
5. In verilated_vpi.h support the VPI calls accessing the parameters.

elgorwi, 2020-05-22:
OK, thanks for the info. I'll look into this.

elgorwi, 2020-05-24:
Made a simple test, starting from t_vpi_memory.v adding some parameters to set the WIDTH and DEPTH of the memory (both with /* verilator public_flat_rd */) and added a vpi_handle_by_name() call to the .cpp for the WIDTH and DEPTH parameters.

When running the test it looks like the WIDTH and DEPTH handles cannot be found. varFind is returning null:
https://github.com/verilator/verilator/blob/50662751fe711cc3f46a21646d855cbfa3ea2462/include/verilated_vpi.cpp#L1096

I think this line could be the cause of the missing WIDTH and DEPTH handles:
https://github.com/verilator/verilator/blob/50662751fe711cc3f46a21646d855cbfa3ea2462/src/V3EmitCSyms.cpp#L342

Removing the line above results in an error when the test is getting compiled:
```
In file included from Vt_vpi_param__ALLslow.cpp:3:
./Vt_vpi_param__Syms.cpp:29:48: error: cannot take the address of an rvalue of type 'Vt_vpi_param::_IDatat__DOT__DEPTH'
        __Vscope_t.varInsert(__Vfinal,"DEPTH", &(TOPp->t__DOT__DEPTH), VLVT_UINT32,VLVD_NODIR|VLVF_PUB_RD,1 ,31,0);
                                               ^ ~~~~~~~~~~~~~~~~~~~
```
This is probably related to the comment that was added to the line above:
```// The VPI functions require a pointer to allow modification, but parameters are constants```

Still figuring out how everything fits together, but I welcome some more suggestions on how to proceed.

Thanks

wsnyder, 2020-05-24:
Perhaps add a new varInsert flavor that takes a constant pointer and casts it away internally?

The appropriate vpi write routine should check if it's a parameter and suppress updates if so (be sure to add a test for this).

elgorwi, 2020-05-25:
Still a bit stuck with the error ```cannot take the address of an rvalue```. 
I'm guessing we don't want to pass by value here, as the parameter could be of any type, not just int.
How would we create a pointer to the constant? Do you have a code snippet that you could share?
Thanks

wsnyder, 2020-05-25:
Can you point to your repo or put up a pull request with what you have so far?

elgorwi, 2020-05-26:
I've opened PR: #2370

closed 2020-06-12 by elgorwi