verilator/verilator #2345: Possible docker alignments between buildenv and run

wsnyder, 2020-05-20:
@wallento looking at using buildenv docker (thanks) and doing "diff -u buildenv/Dockerfile run/Dockerfile" suggests some minor differences & maybe they should be closer?

```
-    && DEBIAN_FRONTEND=noninteractive \
-       apt-get install --no-install-recommends -y \
+    && apt-get install --no-install-recommends -y \
```

Might as well match.

```
+                        ccache \
+                        libgoogle-perftools-dev \
```

Should be in both.

```
+ARG REPO=https://github.com/verilator/verilator
+ARG SOURCE_COMMIT=master
-RUN git clone https://github.com/veripool/vcddiff.git && \
+RUN git clone "${REPO}" verilator && \
```

Might as well do similarly.

```
-ENV VERILATOR_AUTHOR_SITE=1
+WORKDIR /work
```

Might as well have both in both - not having the WORKDIR in buildenv is how I noticed these deltas.

Thanks

wallento, 2020-05-20:
Thanks, you are right, I will look into it. They have both grown in parallel, nice that they have converged similar anyhow :)

wsnyder, 2020-05-26:
If you get a chance could you look at this & also why docker seems so astable (maybe their fault?) Thanks

wallento, 2020-05-26:
I believe those are issues with dockerhub, as there are also no logs etc. when it fails sporadically. Restarting the exact same build works.. Will drop them a support request, and also look into alternative building methods.
Sorry, will also get to that issue asap, unfortunately my TODO list has become a stack, but this is the top now :)

wallento, 2020-05-26:
Done as you pointed out, except:
- REPO and SOURCE_COMMIT are supplied at runtime to the buildenv because that is what therun actually is, in the run container the runtime instead supplies the readily built binary, this it needs those. It could be added to buildenv, but this would be a duplication of the default value and I think it makes more sense not to have default values for variables in a different file if possible.
- VERILATOR_AUTHOR_SITE is not needed in the run cotainer as this is supposed to be what a regular user build is

wsnyder, 2020-05-26:
Makes sense & thanks, feel free to push on pass.

commit 0dcca4a9 (2020-05-26): [docker] Unify docker files

Unify the actions in the Dockerfiles between the runtime environment
and the run containers.

Fixes #2345
PR #2371 (2020-05-26): [docker] Unify docker files
Unify the actions in the Dockerfiles between the runtime environment
and the run containers.

Fixes #2345
commit 723d407a (2020-05-26): [docker] Unify docker files (#2371)

Unify the actions in the Dockerfiles between the runtime environment
and the run containers.

Fixes #2345
closed 2020-05-26 by wallento