verilator/verilator #2346: signal SIGSEGV: invalid address in final block

mikeyangsiv, 2020-05-21:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

We see this failure failure when we grew the number of rows in a similar array/function from 2048 to 8192.  Not understanding the root-cause, I had to put a much larger temp array in the shrunk-down example.

Shrunk-down example ...

```systemverilog
module static_array_in_func (
    input bit reset,
    input bit clock
);
      
      typedef bit [127:0] row_t;
      
      function automatic void foo ();
        row_t ary [2**19];  // passes with 2**18 rows
        $writememh("somefile", ary, 0);
      endfunction: foo

      initial begin
        $finish();
      end

      final begin
        foo();
      end
      
endmodule: static_array_in_func
```
(this code works here ... https://www.edaplayground.com/x/3x33)

with that code and some boilerplate C-main code, 
compiled with ...
```sh
$ verilator --cc --exe -sv --assert --vpi --trace --top-module static_array_in_func ./testcase.sv verilator-main.cpp -CFLAGS "-std=c++14"
```

I get ...
```
$ lldb -- obj_dir/Vstatic_array_in_func +random_seed=100
(lldb) target create "obj_dir/Vstatic_array_in_func"
Current executable set to 'obj_dir/Vstatic_array_in_func' (x86_64).
(lldb) settings set -- target.run-args  "+random_seed=100"
(lldb) run
Process 17891 launched: '$DIR/obj_dir/Vstatic_array_in_func' (x86_64)
INFO(../verilator-main.cpp): Random seed is 100
- testcase.sv:15: Verilog $finish
Process 17891 stopped
* thread #1, name = 'Vstatic_array_i', stop reason = signal SIGSEGV: invalid address (fault address: 0x7fffff7fd458)
    frame #0: 0x0000000000435df8 Vstatic_array_in_func`Vstatic_array_in_func::_final_TOP(vlSymsp=<unavailable>) at Vstatic_array_in_func.cpp:90
   87       } while (VL_UNLIKELY(__Vchange));
   88   }
   89  
-> 90   void Vstatic_array_in_func::_final_TOP(Vstatic_array_in_func__Syms* __restrict vlSymsp) {
   91       VL_DEBUG_IF(VL_DBG_MSGF("+    Vstatic_array_in_func::_final_TOP\n"); );
   92       // Variables
   93       WData/*127:0*/ __Vtask_static_array_in_func__DOT__foo__0__ary[524288][4];
(lldb) bt
* thread #1, name = 'Vstatic_array_i', stop reason = signal SIGSEGV: invalid address (fault address: 0x7fffff7fd458)
  * frame #0: 0x0000000000435df8 Vstatic_array_in_func`Vstatic_array_in_func::_final_TOP(vlSymsp=<unavailable>) at Vstatic_array_in_func.cpp:90
    frame #1: 0x0000000000435f8c Vstatic_array_in_func`Vstatic_array_in_func::final(this=0x000000000066b3c0) at Vstatic_array_in_func.cpp:127
    frame #2: 0x0000000000404cd0 Vstatic_array_in_func`main(argc=2, argv=0x00007fffffffd968, env=0x00007fffffffd980) at verilator-main.cpp:160
    frame #3: 0x00007ffff7495830 libc.so.6`__libc_start_main(main=(Vstatic_array_in_func`main at verilator-main.cpp:48), argc=2, argv=0x00007fffffffd968, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffd958) at libc-start.c:291
    frame #4: 0x0000000000404379 Vstatic_array_in_func`_start + 41
(lldb) quit
```

What 'verilator --version' are you using?  Did you try it with the git master version?

I tested this on 4.028 and 4.034.  No, I have not tried this on `master`.

Would you be willing to try to fix Verilator yourself with assistance?

I am not fluent in C/C++, so I am not comfortable with trying to fix this at this time.

wsnyder, 2020-05-21:
Works fine for me on master, but don't think anything changed there.

Your array needs > 8GB of space. Most likely you are running out of memory; Linux automatically sparse allocates so it won't necessarily fail on the allocation.

Another alternative is to use an associative array.

wsnyder, 2020-05-21:
BTW "vmstat" or "top" in another window should help see if you are running out.

mikeyangsiv, 2020-05-21:
I'm running on a server with 128GB of RAM and 13G of swap, so I don't think I'm running out of RAM.

I can try to compile `master` and rerun my testcase.

Any other hints at how to narrow down the problem?

wsnyder, 2020-05-21:
Does "ulimit" show unlimited?  Also batch systems also have process limits.

Does it fail without --threads?  Without --vpi?  Remove flags to get to minimum.

When you backtrace compare the pointers to the address of the "ary" array, does it make sense?

mikeyangsiv, 2020-05-21:
`ulimit` shows `unlimited`, but `ulimit -s` shows `8192`.  When I increase the stack limit, then this case passes.

wsnyder, 2020-05-21:
Great.

closed 2020-05-21 by wsnyder

mikeyangsiv, 2020-05-21:
Just for comparison, we don't have a stack-size issue with a recent version of the commercial simulator that we also use.  I guess there would be a performance trade-off in allocating this on the heap instead of the stack.

And with a better understanding of the issue, I am able to reproduce this with the original 8192 rows, but with multiple `final` blocks, which is again modulated by `ulimit -s`.

wsnyder, 2020-05-21:
Yes, putting it on the stack is potentially enormously faster. Also if you have a bunch of foo()'s you'd need that memory times the number of foo()s (or not, depending how smart the optimizations are).

mikeyangsiv, 2020-05-21:
Looking into the history of the Linux default stack size (8MiB) ... it seems like that is generally considered "generous".

If the `final` blocks are called serially, then they do not all need to be on the stack simultaneously, right?  Is there a compile-time optimization that is inlining the calls to the `foo()`s?

wsnyder, 2020-05-21:
Correct.