verilator/verilator #2348: question about BLKANDNBLK

RaynardQiao, 2020-05-22:
in test_regress/t/t_altera_lpm.v  line 5753 , module lpm_fifo_dc_dffpipe,  Contains the following code:
always @(posedge aclr or posedge clock)
begin
  if (aclr)
  begin
    for (i = 0; i <= lpm_delay; i = i + 1)
        dffpipe[i] <= 0;
    q <= 0;          //  mark1:  NBLK assign
  end
  ...........
end

always @(d)
begin
   if (lpm_delay == 0)
       q <= d;  // mark2: BLK assign 
end

question is :   mark1 and mark2.    If the parameter lpm_delay is set to 0， I think the tool should report an error Error-BLKANDNBLK， But the current tool has report nothing，Is this correct？

wsnyder, 2020-05-22:
I believe this is correct.

It's not a MULTIDRIVEN because lpm_delay selects either one always or the other always, both never drive.
It's not a BLKANDNBLK for both the reason above, and also because it is always assigned using nonblocking assignments not a mix of blocking and nonblocking assignments.

Agreed?

RaynardQiao, 2020-05-26:
ok, I got your point, thanks

closed 2020-05-26 by RaynardQiao