verilator/verilator #2349: Q:  could verilator specify dump hierarchy ?

jijingg, 2020-05-23:
For some large Modules, dump the whole hierachy results in a very large waveform size, and also slows down the simulation speed.  Even though I know that there have method to  specify a dump period, it still cannot meet thedemand.
so is there any way to dump the hierachy which i care about, or switch off which i dont care, something like this:
```verilog
initial begin
    $fsdbDumpfile("./turbo_tb.fsdb") ;
    $fsdbDumpvars(0,tb.gpsTop.ctrl);           //open dump on this hierarchy
    $fsdbDumpvars(2, tb.gpsTop.corrlator.ctrl);//open dump on this hierarchy
    $fsdbDumpOff(tb.gpsTop.corrlator.vau);     //close dump on this hierarchy
    $fsdbDumpflush;
end    
```

gezalore, 2020-05-23:
The only method I'm aware of is to use the tracing_off/tracing_on mechanism either in source or in the configuration file. This is described in the manual (https://www.veripool.org/wiki/verilator/Manual-verilator) and you can find an example in this test: https://github.com/verilator/verilator/blob/master/test_regress/t/t_trace_ena.v 

I also recommend you make sure you are using the latest version of Veraiator as there have been large tracing speed improcements going in recently. If the dump size is of an issue, you can consider using FST tracing instead (`--trace-fst`) which yields many times smaller files than VCD.

jijingg, 2020-05-23:
@gezalore,  thank you very much, the tracing_off/tracing_on is work.
and yes , the latest version V4.0.34 have 30%~40% speed up than V4.006 on VCD,
it seems more large speed up on FST  👍

gezalore, 2020-05-23:
Great to hear you are getting decent speed gains! Sounds like you are happy with with the solution so closing this one.

closed 2020-05-23 by gezalore