verilator/verilator #2357: Usability: Support building an AppImage

wallento, 2020-05-24:
Verilator is still constantly improving and adding great features, and is getting increasingly supported by virtually all open source IP, there still is the issue that people may be offset by having to build verilator regularly :) Also, adding a particular version to various CI flows is a bit of an "issue".
The run-docker container was a first approach, but it suffers from a couple of issues (docker requirement may be bad for CI, not arbitrary files in the system can be used).
I therefore had a brief look into portable alternatives. The most widespread are snap, flatpak and AppImage. I have no expertise on those, but AppImage seems a decent candidate. Shortly summarized it packs a portable installed version of a program into a ramdisk, puts that into an elf, which then at runtime fuse-mounts this into /tmp. This works quite well.
There are a couple of things to check/implement:

- [ ] I assume we need the ability for a static (fat) build
- [ ] An installation must be relocatable
- [ ] The runtime library must be copied into the generated output. The docker image does this too, will open an issue to support this generally as I think it has good value beyond this issue.
- [ ] Bring up AppImage build
- [ ] Continuous deployment of AppImage, so that people can use automatically generated builds

Especially for the first two I am halfway confident that they are mostly done, but I will now start playing around.
Do you like the idea, @wsnyder?

wallento, 2020-05-24:
`--disable-defenv` is relevant in this context for relocatable installations. AppImage needs to set the environment path to its `APPDIR`

wsnyder, 2020-05-24:
The sites I'm aware of including mine just have multiple versions of verilator checked out and point VERILATOR_ROOT at the right one, doing that should be fully relocatable today.  This also works with all other tools including commercial simulators, so seems likely to be the model for quite some time with such sites.  So, I'm not sure how much the packaging stuff will help or be wanted by those doing that now, but I don't have any objection to trying this out experimentally.

Snap comes installed with Ubuntu 20.04 LTS which gives it points, but doesn't seem as portable as app image.

wallento, 2020-05-25:
It was quite simple to get a simple mockup running: https://github.com/verilator/verilator/tree/appImage
Build instructions:
```
wget https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage
APP_IMAGE_TOOL=~/Downloads/appimagetool-x86_64.AppImage ./build-appImage.sh
```
And then to use it copy `Verilator-x86_64.AppImage` whereever you want as replacement for the `verilator` command or even rename.

I agree it is not about the sites, but I think this can be pretty handy for various travis jobs or quickstarts.

wsnyder, 2025-02-25:
Going to close this as unplanned as unlikely to get worked on, if someone wants to contribute a patch it might be accepted - reopen and discuss here.

closed 2025-02-25 by wsnyder