verilator/verilator #2358: Add switch that copies and emits local copy of library files

wallento, 2020-05-24:
Currently, the Makefile (and cmake file) refer to the library files as found in VERILATOR_ROOT (which is set in the Makefile).
There are a couple of possible reason why this may not be desirable:
- The user has multiple version of Verilator and just messes things up
- The user wants to distribute the files without requiring a (matching) installation of Verilator
- AppImage or similar happens (#2357)
The problem is currently solved in the docker run container by copying the runtime files to the output directory and rewriting VERILATOR_ROOT: https://github.com/verilator/verilator/blob/master/ci/docker/run/verilator-wrap.sh#L24
A better way is to add a switch (proposal: `--Minclude-runtime` or so?) that:
- Copies the library files in the emit phase to `vlt-lib`
- Emits `VERILATOR_ROOT=./vlt_lib`

gezalore, 2020-05-24:
Hey Stefan, long time no see! If you don't mind me pitching in I would like to add some thoughts.

I wonder, are there any use cases where what you are proposing is not beneficial? (i.e.: should we always just emit the necessary run-time libarary files to the obj_dir?) 

> The user wants to distribute the files without requiring a (matching) installation of Verilator

I find this comes up in conversation every now and again, often in a form of wanting to compile the model on a different host. In my understanding you indeed need a matching installation of Verilator but actually I think the situation is a bit more murky as we also depend on include/verilated.mk, which in turn depends on the platform discovery performed by configure so at least from a theoretical point of view you can have issues. In partiocular, all the various -W and -f compiler flags for example are baked into verilated.mk at configure time, does AppImage package the compiler as well? Otherwise I think there will be trouble. Maybe this does not cause too much headace in real use, but I wonder if it can be reconciled at the same time? Essentially this would require the Verilator output directory to then be fully independent of the installation (i.e.: no more VERILATOR_ROOT).

On another note, while I'm no lawyer, I wonder if copyting the run-time library could create licensing issues as if you just distributed the output of Verilator, it would now contain LGPL components, while previously it only conatined generated output?

wsnyder, 2020-05-24:
At present the libraries and generated code are assumed to match up, there is no effort to allow multiple versions of verilator to be present. (Really there should be an API version number that throws up if they are incompatible.) Note there was also some discussion in  #2041.

If you are using the library do you really need the verilated makefiles at all?  Is this to get the switches?  It seems odd to both want the verilated makefiles but not have a VERILATOR_ROOT.

The intent of the LGPL is to allow linking regardless of where the files are from, I don't see why this proposal would matter to that.

gezalore, 2020-05-24:
> At present the libraries and generated code are assumed to match up, there is no effort to allow multiple versions of verilator to be present.

Agreed and the library is currently "matched up" by pointing VERILATOR_ROOT to the right palce when bulding the model. I think the problem that Stefan is describing is that in the case of Docker or AppImage, the place where VERILATOR_ROOT should point to is ephemeral (only lives during the life of the container, which terminates when verilation is finished, before the model is built), so where should it point to during build of the model?

> If you are using the library do you really need the verilated makefiles at all? Is this to get the switches?

I think you would still like to have the niceness of having a build system come with your verilated model that "just works".

> The intent of the LGPL is to allow linking regardless of where the files are from, I don't see why this proposal would matter to that.

As I mentioned I am no expert just paranoid  so I'm happy no to discuss this further, I just wanted to make sure the implications are udnerstood. In my understanding the model indeed would still be "software that uses the library" rather than derivarive work even though it's distributed in the same bundle.

gezalore, 2020-05-24:
> I think you would still like to have the niceness of having a build system come with your verilated model that "just works".

.. even if you verilated it via the Docker/AppImage distribution that is.

wsnyder, 2020-05-24:
BTW I understand the desire of having a library all-contained, the train wreck is once people have two of those libraries made with different versions, or their app uses a different version than the library.  Part of the mess is DPI/VPI calls and threads that are required to have a holistic view of the model and cannot all simply be state that is only per-sub-model.

gezalore, 2020-05-24:
I think we are in agreement but maybe looking at different parts of the problem. Let me try to list the components of this issue as I see it in an attempt to present the whole picture:

1. Generated code depends on the interface of the run-time library. This can be checked by the API version mechanism you mentioned to at least get an error (if it links at all). 

2. We have global singleton state, which needs to be aware of all models in the app, including ones verilated separatel. This I believe is the bit about 'having holistic view of the model' you mentioned above. 

Taking point 1 and 2 together precludes using models created by differente versions of Verilator in the same app as there can only be one version of the singletons. This I would think is OK, and we definitely do not want to go down the route of having models compatible with multiple versions of the run-time library. Also, I think this is same train wreck you mentioned above albeit happening at a different station on the train line.

3. verilated.mk provides you the nice build system for the model but compiler switches and stuff depends on configure. 

4. When packaging Verilator via Docker (and I presume AppImage?), VERILATOR_ROOT is ephemeral, so where do we get the library and verilated.mk from if we just try to build the verilated model on the same host? Currently we copy it out of the docker container, but if the environment (say compiler version) on the host is different than in the container, it will likely not work due to point 3.

5. Not immediately relevan but related, is that the run-time lib compilation actually also depends on verilator options making models compiled with different options potentially incompatible.

I think Stefan meant to address point 4 with this issue. I am just trying to highlight that it's problemetic due to point 3 as well, and you are rightly highlighting point 1 and 2. I appologize if I am running around in circles, and if this is all clear. I am not trying necessarily push the single all-contained library again, just trying to make everything work if possible :) 

BTW I think it would be possible to avoid the trainwreck caused by point 1 and 2 using only standard linkage semantics by linking the run-time library into every "sub-model", and at [no later than] run-time detecting that all sub-models were linked with the same version of the lib, so it does not matter which particualr sub-model provided the  run-time lib, its all the same. (Might not be the cleanest but it can be done.)

In essence this problem is similar to what GCC has with libgcc and libc which I think they solve with library/symbol versioning or multilibs but there is no way around incompatible global state.

I'm sorry I didn't mean to hijack this one. The immediate question in this issue is, do we want to be able to support packaging Verilator in ephemeral containers (or similar), while providing the exact same functionality (e.g.: being able to use the canned makefile) as a local installation? If so point 4 above and dependent problems need _some_ solution.

wsnyder, 2020-05-24:
Great summary.
1 & 2: I generally agree the way to address this would be to medium term (or otherwise) split verilated* into two flavors - those that are inlines and such and are only used in a model (so could change without link problems) and those that need a stable API and should be asserted to match.  Will file a bug for discussion.
4: As above I don't personally have this problem and think many/most larger sites don't either, but I can appreciate it is a problem for some, and welcome the experimentation.

wallento, 2020-05-24:
@gezalore Hey, Geza! Already saw your name pop up here, awesome!

I also don't see the issue about _sites_, they are all managed well (in their own sense sometimes). It is really more about the vast number of Makefiles, READMEs and CI files I saw over the last couple of years in open source repositories that all had their own variant of "git clone verilator and build it at this version". Verilator has become the de facto standard for many projects, but thanks to @wsnyder's incredible productivity people pretty regularly build Verilator for yet another RISC-V core they want to try out :)

I didn't think much about the discussed points before, because the majority of cases I had in mind don't really suffer from them I believe. Anyhow, you guys are right, definitely more discussion needed.

wallento, 2020-05-24:
A very rough mockup of copying some files: https://github.com/wallento/verilator/tree/copy-runtime-files
I will use it as base for some tests, especially portability and the points discussed.

gezalore, 2020-05-24:
One obvious thing I would expect to blow up with this is if you try to build the output on a host which has no ccache on the PATH as we now bake that in from configure. Sure you can set "OBJCACHE=" (empty), but I'm not sure the cognitive load is less than having to install the local copy. BTW I do see value in having an "all-you-need-is-internet" way of getting various versions of Verialtor, I myself have gone throught he chores you mention about CI or cloud platforms. I have also seen _sites_ with shall we say less structured compute infrastructure with each host running slightly different versions of OSs with no standard tool installations so I am all for the idea :).

wsnyder, 2022-01-07:
BTW note that ci/docker/run/verilator-wrap.sh currently makes a "vlt" directory.

GeorgeLyon, 2022-01-07:
I ran into this issue recently as well when using a docker image of Verilator and then compiling on the host machine (hence the issue above, and the slight bug that copying VERILATOR_ROOT only works when you do not specify `-Mdir`).

--- 

From @gezalore 
> We have global singleton state, which needs to be aware of all models in the app, including ones verilated separatel. This I believe is the bit about 'having holistic view of the model' you mentioned above.

I'm not an expert in such matters, but I do think it is possible, and there is a use case for having multiple verilated models in the same app. For instance, you can have a number of separate circuits described in an HDL, that you wish to unit-test independently (and possibly in parallel). I'm hoping to achieve this by compiling a dynamic library, ensuring the all symbols are hidden with the exception of the few defined in a harness (you currently have to trick Verilator a bit to achieve this, but [Chisel does the same thing](https://github.com/ucb-bar/chiseltest/blob/master/src/main/scala/chiseltest/simulator/jna/JNAUtils.scala)). Then, you can happily `dlopen` the library, `dlsym` your harness symbols, and `dlclose` when you are done. This **should** stand up to running multiple Verilated models at the same time, as their singletons should be initialized to different memory. 

--- 

Overall it would be great to be able to have Verilator create mostly-portable code, and given the discussion in this thread it seems `make` may not be the best tool for this job. Perhaps we can achieve what we want by having a lower-level build tool (specified via the `--make` option) which emits all necessary C++ sources and headers to the output directory (we can include flags like VM_THREADS in a single header file) which users can then hook up to their own build system of choice. This would also enable working around limitations like `make`'s patch support for paths-with-spaces.