verilator/verilator #2359: Discussion on fracturing verilated runtime libraries for packaging and protect-lib

wsnyder, 2020-05-24:
Discussion for now but likely eventually a proposal for feature.   CC @gezalore @wallento @toddstrader 

#2358 and earlier #2041 discussed issues around supporting "libraries" of verilated models.

Ideally a verilated model could be compiled into a library, and that library could be used anywhere, even mixed with verilated models from other versions of Verilator (protect-lib ideally wants this).

The problem is that
1. Currently Verilator options change how the verilated code is built, and
1. There is shared state between the models, e.g. DPI, VPI, tracing, coverage, threads. As this code is common currently the same include/ code must be used in all verilated model libraries that are built.

Here's some ideas for discussion:

1. Note soon we will be able to require C++11, and once that is in place some of the compilation flag-caused differences will be removed (e.g. related to classes/queues being used.)
1. Threading use may have a significant performance overhead.  Perhaps a runtime flag to bypass the locks etc would have close enough performance to non-threaded be considered.  Perhaps also there is some binary patchup that could NOP the locks similar to how the kernel does it.  Research needed.
1. The verilated* calls would be split into flavors,
   a. those that are inlined and as such if different models use different functions it won't matter.
   b. those that are not inlined but "pure" in that different models after linking into an archive could resolve the different versions away.
   c. those that truly need global state. (perhaps called through shims from code of flavor a or b)
1. The global state would need a stable API.
1. Whatever global API calls remain there should be global or per-feature min/max API version identifiers compiled into both the caller and library which are checked in the library at construction time. Any API incompatible change that the library couldn't shim or backwards-support would be reported as an error. 
1. It is probably possible and desirable once the above is in place to build a single libverilated.a with the compete global runtime library.

wsnyder, 2020-05-24:
While not the important point to discuss, I found pthreads already has code to skip the mutex when not threaded, though it's still slow and better to branch around the mutex or avoid entirely: https://stackoverflow.com/questions/32317370/avoid-cost-of-stdmutex-when-not-multi-threading

gezalore, 2020-05-24:
Just to add to the TODO list, for the skipping the mutex part the GNU IFUNC resolution mechanism could work too. This is basically how they can swap in AVX versions of libc routines at runtime when you have AVX, but the SSE version when you don’t. It uses a similar or same PLT mechanism as shared libraries do, so you could resolve to an empty function without threads. Might be faster, though not as portable.

toddstrader, 2020-05-26:
I know we've discussed runtime compatibility vs runtime encapsulation before in #1518 and elsewhere.  However to recap, --protect-lib currently links the runtime into the library it is building.  This is fine (?) when simulating with Verilator because only one copy of the runtime symbols is left in the end and presumably they are the same or compatible.  It is necessary for running with other simulators as they clearly don't have the Verilator runtime.

This is why I initially suggested wrapping the runtime in a namespace for each --protect-lib library.  It would enable us to sidestep the compatibility problem and I believe that not sharing trace information with each other and potentially the root Verilator runtime is the correct default behavior for --protect-lib.

However, --protect-lib appears to being used more for incremental builds than for IP protection so we definitely need to handle both cases.  Given that @wsnyder 's bullets seem good given that:
1. We can still run --protect-lib libraries with non-Verilator simulators (t_prot_lib* worked with XSim the last I checked)
2. You have to opt into global state sharing when using --protect-lib
3. We strive for API stability and backwards compatibility so that protected libraries and Verilator are unlikely to conflict with each other

wsnyder, 2020-05-26:
There are still common routines at play which a namespace won't help.  For example if the models do any $fopens.  Even if we refactor to make such details local to a given model rather than global (which we should consider as part of this), multithreading by definition needs to have a shared pool as you can't have every model each wanting all the CPUs ;)

Also I believe tracing would generally want to always be global.  Getting a separate trace from each IP instantiation isn't a pleasant or desirable user experience.

wsnyder, 2020-05-26:
I think changing from enum { PARAM_NAME = value } to a "const dtype PARAMNAME= value;" for simple types should allow both taking the address, and also using it to declare type widths.  For complex types we might need to drop the const so the constructor can build them appropriately.

gezalore, 2020-05-26:
Did you mean for the last one to go in #2344?

wsnyder, 2020-05-26:
Right, that comment was for the other issue, will delete it here.

closed 2024-04-13 by wsnyder