verilator/verilator #2360: Should initial, settle and final blocks be in Slow files?

gezalore, 2020-05-24:
I am under the impression that these are only used during the very first and very last time steps per simulation. Yet, the settle block is 40% of the generated model code in SweRV EH1. Should we stick these in __Slow.cpp files to improve compilation speed? Note that this would mean always having an __ALLslow.cpp file when building without VM_PARALLEL_BUILDS, but hopefully given that that file is compiled with -O0 it would not add a lot to build times of small things?

wsnyder, 2020-05-24:
Initial/settle should already be slow files. 

            if (domainp->hasInitial() || domainp->hasSettle()) newFuncpr->slow(true);

However sometimes the V3Combine step realizes these functions are identical to fast path functions and they get promoted to fast (because they then are). The way to check is if they are called in the normal eval() or just run a profile and look at the count.  Certainly could be a bug here...

gezalore, 2020-05-24:
Hmm, seems like the bug option to me. The only call I see is in _eval_settle which is the one called once at the beginning via _eval_initial_loop I believe. Same goes for EH2. I'm happy to investigate a bit later unless you fancy.

```
[gl@blackbox build-a] % grep -R -C7  _settle__TOP__2 obj_dir/*.cpp
obj_dir/Vtb_top.cpp-            __Vtask_tb_top__DOT__preload_iccm__4__addr
obj_dir/Vtb_top.cpp-                = ((IData)(4U) + __Vtask_tb_top__DOT__preload_iccm__4__addr);
obj_dir/Vtb_top.cpp-        }
obj_dir/Vtb_top.cpp-        __Vlabel2: ;
obj_dir/Vtb_top.cpp-    }
obj_dir/Vtb_top.cpp-}
obj_dir/Vtb_top.cpp-
obj_dir/Vtb_top.cpp:void Vtb_top::_settle__TOP__2(Vtb_top__Syms* __restrict vlSymsp) {
obj_dir/Vtb_top.cpp:    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtb_top::_settle__TOP__2\n"); );
obj_dir/Vtb_top.cpp-    Vtb_top* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
obj_dir/Vtb_top.cpp-    // Variables
obj_dir/Vtb_top.cpp-    CData/*3:0*/ __Vtableidx1;
obj_dir/Vtb_top.cpp-    WData/*95:0*/ __Vtemp42[3];
obj_dir/Vtb_top.cpp-    WData/*95:0*/ __Vtemp46[3];
obj_dir/Vtb_top.cpp-    WData/*95:0*/ __Vtemp50[3];
obj_dir/Vtb_top.cpp-    WData/*127:0*/ __Vtemp70[4];
--
obj_dir/Vtb_top.cpp-
obj_dir/Vtb_top.cpp-void Vtb_top::_eval_settle(Vtb_top__Syms* __restrict vlSymsp) {
obj_dir/Vtb_top.cpp-    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtb_top::_eval_settle\n"); );
obj_dir/Vtb_top.cpp-    Vtb_top* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
obj_dir/Vtb_top.cpp-    // Variables
obj_dir/Vtb_top.cpp-    IData/*31:0*/ __Vilp;
obj_dir/Vtb_top.cpp-    // Body
obj_dir/Vtb_top.cpp:    vlTOPp->_settle__TOP__2(vlSymsp);
obj_dir/Vtb_top.cpp-    __Vilp = 0U;
obj_dir/Vtb_top.cpp-    while ((__Vilp <= 0x136U)) {
obj_dir/Vtb_top.cpp-        vlTOPp->__Vm_traceActivity[__Vilp] = 1U;
obj_dir/Vtb_top.cpp-        __Vilp = ((IData)(1U) + __Vilp);
obj_dir/Vtb_top.cpp-    }
obj_dir/Vtb_top.cpp-}
obj_dir/Vtb_top.cpp-
```

wsnyder, 2020-05-24:
Please look into it, thanks.

gezalore, 2020-05-24:
Hmm, acutally the answer is more obvious than I expected:
https://github.com/verilator/verilator/blob/f7249ad23a79c0552946a7765c773fb821a85603/src/V3EmitC.cpp#L3776-L3781
without --output-split, it's all just emitted into the same file... Seems like we are back to #2247 But at the same time just disabling that condition above (i.e.: emitting the fast/slow separately, but without forther output split) makes the EH1 build pretty much 100% faster when run in parallel, and abotu 60% when run with -j1. I am surprised the SweRV folks don't use --output-split, but at the same time I know the lowRISC projects also didn't use --output-split for a long time (partly I think because FuseSoC doesn't). What do you think, should we change anything in Verilator? I'm tempted to at lease PR SwerRV to turn on output split..

gezalore, 2020-05-24:
BTW, one thing we could do to speed up small builds is to change verilated.mk to create only a single __ALL.cpp file when OPT_FAST is the same as OPT_SLOW. Doing that + always emitting the separate slow file might give us most of the good things of both worlds?

wsnyder, 2020-05-24:
My paper (https://www.veripool.org/projects/veripool/wiki/Papers) certainly suggested output split should be on.  No objection to suggesting that to them.

However I think we discussed making output-split on by default which is probably a better average user experience.  I want it to be turned on only when over a certain size threshold, otherwise we're just wasting file IO and compiler time as the headers being read twice will be more expensive than anything saved (e.g. the test_regress suite mentioned last time).

gezalore, 2020-05-24:
> However I think we discussed making output-split on by default which is probably a better average user experience. 

We did discuss this and I couldn't agree more.

>  I want it to be turned on only when over a certain size threshold, otherwise we're just wasting file IO and compiler time as the headers being read twice will be more expensive than anything saved 

I think the inpass is over what magic number to pick? What is optimal with regards to file IO will depend on the host hardware a lot, but also the compiler, so I think trying to evaluate this with benchmarks is not a great use of our time (+ finding various size benchmarks is hard to start with). Also, if we need to go over a certain size to trigger some particular behaviour it makes it harder to test + there might be oddity when suddenly crossing the threshold changes behaviour. Do you have any suggestions on how we can make progress on this? Maybe just `$(wc -l verilated.h) -ge $(wc -l obj_dir/*.cpp)`?

As for the regression suite, I think it does use OPT_FAST == OPT_SLOW most of the time, so the change I suggested above would save that.

wsnyder, 2020-05-25:
I ran some experiments (with ccache disabled) on regressions + extended tests normally.  This is with the default regression's OPT_FAST=OPT_SLOW=none.

Then I ran the same tests with --output-split 15000 (the number SweRV liked).  10000 might be a good default.

| Suite/case | Wall time | Total compile time (sec) | files compiled | Crit path (sec) |
| --- | --- | --- | --- | --- |
| test_regress baseline | 6:06 | 6143 | 7400 | 3405 |
| test_regress split    | 7:04 | 7214 (117%) | 9910 (133%) | 2978 (87%) |

So, we've added a minute of wall time, because the compiler has 17% more total compiling to do and 33% more files to do.  The last measurement is synthetic - if we theoretically ran each test separately so we could be limited by the longest compile for that test - similar to what a user would see compiling one model - then the splitting is getting the user what they want 13% faster.

Obviously I don't want to add a minute to regressions (and a lot more to Travis with 2 CPUs).

Looked into what's going on, most of the time is because once we turn on split we now make __Syms as a separate file. I hacked that off and got about 1/2 of the extra time back.  The rest of the extra time is because we set VM_PARALLEL_BUILDS when using csplit.  Turning that off helped most of the other 1/2.  However we are still compiling a __ALLfast and __ALLslow with VM_PARALLEL_BUILDS==0.

Changing to use VM_PARALLEL_BUILDS==0 and building a single __ALLfast (no slow):

| Suite/case | Wall time | Total compile time (sec) | files compiled | Crit path (sec) |
| --- | --- | --- | --- | --- |
| test_regress baseline | 6:06 | 6143 | 7400 | 3405 |
| test_regress split    | 7:04 | 7214 (117%) | 9910 (133%) | 2978 (87%) |
| test_regress all      | 5:06 | 5088 (82%) | 4872 (65%) | 3007 (89%) |

Score!  In a real implementation we can set this to 1 if any splits actually occur. Only 7 tests do this in the regressions now.

I ran other experiments with using -Os and got similar results.

I ran experiments with extended tests with -Os and it of course likes output-split:

| Suite/case | Total compile time (sec) | files compiled | Crit path (sec) |
| --- | --- | --- | --- |
| extended -Os base  | 309        | 14        | 305        |
| extended -Os split | 334 (108%) | 74 (524%) | 193 (63%)  |

A clear win to add 8% computrons for a 40% reduction in observed critical path.

So potential suggested actions:
- [ ] Set VM_PARALLEL_BUILDS=1 only when splitting occurs  (Separate patch)
- [ ] make __ALL instead of __ALLfast/__ALLslow when VM_PARALLEL_BUILDS=0 (Separate patch)
- [ ] Want to run any experiments to determine split amount? Otherwise suggest 10000.
- [ ] Fix some (possibly test) bugs seen when tried --output-split always on
- [ ] --output-split 10000 default

Want to tackle any of these?

gezalore, 2020-05-25:
This looks great, I will get right on it.

gezalore, 2020-05-25:
Would you mind double checking that #2365 ends up on the last line in your middle table (test_regress all) please? Travis build speeds looked encouraging.

wsnyder, 2020-05-25:
4:43. Looks great.

gezalore, 2020-05-26:
Done in #2365

closed 2020-05-26 by gezalore