verilator/verilator #2361: vpi attributes should respect Verilog-2001 attributes

sequencer, 2020-05-25:
Here is the sample example code from documentation:
```
module our (input clk);
  reg readme /*verilator public_flat_rd*/;
  initial $finish;
endmodule
```
If I use [standard attributes](https://link.springer.com/chapter/10.1007%2F978-1-4615-1713-9_27) like this:
```
module our (input clk);
  (* verilator public_flat_rd *)
  reg readme;
  initial $finish;
endmodule
```
It will raise an exception:
```%Error: ../sim_main.cpp:10: No handle found```

So here is the problem that:
the attributes of verilator is not a standard ieee implementation, which will cause verilog generator(like chisel) to implement a dedicated attribute implementation to verilator.

Hope verilator can implement a standard attributes spec.

wsnyder, 2020-05-25:
Attributes have always seemed like one of those good ideas , but in practice metacomments are still much more popular.

Anyhow I have no objection to additionally supporting metacomments for those attributes that can appropriately use them (e.g. attaching to signals etc) versus e.g. warning disables which are arbitrary-token based.

Perhaps you would be willing to work on a pull request and tests for this?

sequencer, 2020-05-25:
Sure! Although I have no previous experience to verilator codes, but I'm willing to do so, would mind giving some advice to me for what codes should I look into.

wsnyder, 2020-05-25:
1. First see https://github.com/verilator/verilator/blob/master/docs/internals.adoc
1. Write some test_regress tests using attributes in all the legal places.  Test with icarus or another simulator.
1. Should pass on Verilator as will ignore them
1. Change verilator to not strip all attributes and instead parse then in verilog.y
1, Have verilator ignore the parsed attributes doesn't understand.
1. Tests should pass again but now parsed.  Can push at this point.
1. Add new test to use verilator attributes in place of metacomments (add versions of some tests that use e.g. var metacomments)
1. Improve verilog.y to parse them.
1. Tests should pass again but now parsed.  Can push at this point.

sequencer, 2020-05-25:
Sure! Thank you! I'll go deep into those!

wsnyder, 2022-12-16:
I sing due to age; without a pull not likely to happen.

closed 2022-12-16 by wsnyder