verilator/verilator #2362: verilator fails to build with bison 3.6.1

aelmahmoudy, 2020-05-25:
verilator 4.034 fails to build with bison 3.6.1

{{{
ccache g++ -g -O2 -fdebug-prefix-map=/builds/electronics-team/verilator/debian/output/verilator-4.034=. -fstack-protector-strong -Wformat -Werror=format-security -g -O2 -fdebug-prefix-map=/builds/electronics-team/verilator/debian/output/verilator-4.034=. -fstack-protector-strong -Wformat -Werror=format-security -I/usr/include -std=gnu++14  -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -Wdate-time -D_FORTIFY_SOURCE=2 -Wdate-time -D_FORTIFY_SOURCE=2 -I/usr/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"/usr/include\" -DDEFENV_SYSTEMC_LIBDIR=\"/usr/lib/x86_64-linux-gnu\" -DDEFENV_VERILATOR_ROOT=\"/usr/share/verilator\" -Wno-char-subscripts -Wno-unused -c ../V3ParseGrammar.cpp
In file included from ../V3ParseGrammar.cpp:22:
verilog.c:686:62: warning: missing terminating " character
verilog.c:686:62: error: missing terminating " character
verilog.c:687:61: warning: missing terminating " character
verilog.c:687:61: error: missing terminating " character
verilog.c:688:64: warning: missing terminating " character
verilog.c:688:64: error: missing terminating " character
verilog.c:689:66: warning: missing terminating " character
verilog.c:689:66: error: missing terminating " character
verilog.c:690:72: warning: missing terminating " character
verilog.c:690:72: error: missing terminating " character
verilog.c:691:63: warning: missing terminating " character
verilog.c:691:63: error: missing terminating " character
verilog.c:692:67: warning: missing terminating " character
verilog.c:692:67: error: missing terminating " character
verilog.c:693:73: warning: missing terminating " character
verilog.c:693:73: error: missing terminating " character
verilog.c:694:70: warning: missing terminating " character
verilog.c:694:70: error: missing terminating " character
verilog.c:695:68: warning: missing terminating " character
verilog.c:695:68: error: missing terminating " character
verilog.c:696:59: warning: missing terminating " character
verilog.c:696:59: error: missing terminating " character
verilog.c:697:61: warning: missing terminating " character
verilog.c:697:61: error: missing terminating " character
verilog.c:698:67: warning: missing terminating " character
verilog.c:698:67: error: missing terminating " character
verilog.c:699:60: warning: missing terminating " character
verilog.c:699:60: error: missing terminating " character
verilog.c:700:65: warning: missing terminating " character
verilog.c:700:65: error: missing terminating " character
verilog.c:701:68: warning: missing terminating " character
verilog.c:701:68: error: missing terminating " character
verilog.c:702:68: warning: missing terminating " character
verilog.c:702:68: error: missing terminating " character
verilog.c:703:67: warning: missing terminating " character
verilog.c:703:67: error: missing terminating " character
verilog.c:704:63: warning: missing terminating " character
verilog.c:704:63: error: missing terminating " character
verilog.c:1159:70: warning: missing terminating " character
verilog.c:1159:70: error: missing terminating " character
verilog.c:1160:69: warning: missing terminating " character
verilog.c:1160:69: error: missing terminating " character
verilog.c:1161:72: warning: missing terminating " character
verilog.c:1161:72: error: missing terminating " character
verilog.c:1162:74: warning: missing terminating " character
verilog.c:1162:74: error: missing terminating " character
verilog.c:1163:80: warning: missing terminating " character
verilog.c:1163:80: error: missing terminating " character
verilog.c:1164:71: warning: missing terminating " character
verilog.c:1164:71: error: missing terminating " character
verilog.c:1165:75: warning: missing terminating " character
verilog.c:1165:75: error: missing terminating " character
verilog.c:1166:81: warning: missing terminating " character
verilog.c:1166:81: error: missing terminating " character
verilog.c:1167:78: warning: missing terminating " character
verilog.c:1167:78: error: missing terminating " character
verilog.c:1168:76: warning: missing terminating " character
verilog.c:1168:76: error: missing terminating " character
verilog.c:1169:67: warning: missing terminating " character
verilog.c:1169:67: error: missing terminating " character
verilog.c:1170:69: warning: missing terminating " character
verilog.c:1170:69: error: missing terminating " character
verilog.c:1171:75: warning: missing terminating " character
verilog.c:1171:75: error: missing terminating " character
verilog.c:1172:68: warning: missing terminating " character
verilog.c:1172:68: error: missing terminating " character
verilog.c:1173:73: warning: missing terminating " character
verilog.c:1173:73: error: missing terminating " character
verilog.c:1174:76: warning: missing terminating " character
verilog.c:1174:76: error: missing terminating " character
verilog.c:1175:76: warning: missing terminating " character
verilog.c:1175:76: error: missing terminating " character
verilog.c:1176:75: warning: missing terminating " character
verilog.c:1176:75: error: missing terminating " character
verilog.c:1177:71: warning: missing terminating " character
verilog.c:1177:71: error: missing terminating " character
make[3]: *** [../Makefile_obj:298: V3ParseGrammar.o] Error 1
make[3]: Leaving directory '/builds/electronics-team/verilator/debian/output/verilator-4.034/src/obj_dbg'
make[2]: *** [Makefile:66: ../bin/verilator_bin_dbg] Error 2
make[2]: Leaving directory '/builds/electronics-team/verilator/debian/output/verilator-4.034/src'
make[1]: *** [Makefile:221: verilator_exe] Error 2
make[1]: Leaving directory '/builds/electronics-team/verilator/debian/output/verilator-4.034'
}}}

Full build log can be found at: https://salsa.debian.org/electronics-team/verilator/-/jobs/763246/raw

wsnyder, 2020-05-25:
This is believed fixed in the master unreleased version, see #2320 please.

closed 2020-05-25 by wsnyder

vowstar, 2020-06-07:
Please update to bison 3.6.3, or use latest master branch of verilator.

https://lists.gnu.org/r/bison-patches/2020-05/msg00057.html
https://github.com/akimd/bison/releases