verilator/verilator #2366: Optimize ternary expressions with the same condition

gezalore, 2020-05-25:
The code that makes LLVM's Jump Threading optimization explode in time on EL2 (as mentioned in #2365) is of the forms:
```
x0 = cond0 ? then0 : else0;
x1 = cond1 ? then1 : else1;
x2 = cond2 ? then2 : else2;
..
xn = condn ? thenn : elsen;
```
condN is pretty much always the same (reset signal, or reset combined with something else), thenN is an arbitrary expession and elseN is often constatnt 0. Otherwise the statements are largely independent. It think it would be worth trying a pass which rolls these up into:
```
if (cond) {
  x0 = then0;
  ...
  xn = thenn;
} else {
  x0 = else0;
  ...
  xn = elsen;
}
```
I am happy to have a stab at this but would appreciate some suggestion on which point in the pipeline this would be most appropriate, assuming you think this is useful. It might give use some of the gains that #2248 would yield. Also has a good chance of reducing code size incase the C compiler gave up merging the blocks because of complexity. Also if there is a standard mechanism which test if 2 trees (condN and condM in the example) are the same, please point me towards it.

gezalore, 2020-05-25:
Reading the objdump, the compilers do roll some of these up, with various success. Clang 10 takes 50 seconds to copmile a single example and is left with 125 branches, while GCC 9.3 takes 8 seconds and is left with 189 branches.

wsnyder, 2020-05-26:
This should be done after most other optimizations, the V3Reloop optimization is the right place for this - basically it builds up higher level constructs near the end and this smells similar.

Be careful if there's an "x = ....;   y = ... ? x" or similar. If statements stay in same order should be OK, but perhaps will miss too many opportunities if you don't pull conditionals out of the middle of other statements.

closed 2020-05-30 by gezalore