verilator/verilator #2367: Inline short Active blocks into _eval

gezalore, 2020-05-25:
EH1 has a lot of examples where a sequent CFunc contains only a handful ot statements (often 1 or 2) and is called only once. This causes a performance penalty when --output-split puts these in a different function than _eval and hence the compiler can't inline them, so we should inline these ourselve when it's obviously the right thing to do.

wsnyder, 2020-05-26:
Right now the functions are made first, then the wrapper later.  It should be easy to check they are used only once then insert them.  Be careful of any local variables (maybe don't optimize if the inserting function has local variables.)

Riya123-code, 2023-04-29:
hello,
Can you please explain this issue in detail, as i would like to work on it.

wsnyder, 2023-04-30:
Presently V3Sched makes a top function (makeTopFunction), and a function for each clock block (makeSubFunction).

There's several ways to address this.

1. Instead of making a subFunction, build the list of statements first. Then look at the length of those statements (V3InstrCount) and either make the Sub function, or if small enough add to TopFunction directly.
2. Still build the functions as-is, and have a later stage (probably right after V3Combine is called) make an optimization pass that inlines all small functions.  This will offer ferformance wins in areas other than this one, but is more complicated.