verilator/verilator #2368: Split _eval when it grows large

gezalore, 2020-05-26:
Is there any reason we should not make _eval obey --output-split? It is one of the long pole functions in terms of compilation speed on large designs with a lot of sub blocks + if it we do #2367 it has a danger of growing even bigger. Peformancewise I don't think this should hurt much as we can just tail-call the next piece of _eval at the end of the previous, so if the split is not too small this should not be very noticeable.

wsnyder, 2020-05-26:
Agreed, just hasn't been done. Probably cleanest to add another layer of functions when this happens, that is eval() calls eval1() ... eval999().  Will need benchmarking at end to make sure doesn't do something unexpected, which I doubt.

wsnyder, 2020-08-24:
Added splitting for _eval, _eval_settle, _eval_initial, final in master towards version 4.100.

closed 2020-08-24 by wsnyder