verilator/verilator #2369: Add an OPT_GLOBAL to apply to the run-time library

gezalore, 2020-05-26:
As I mentioned in #2365, I think there is a use-case for OPT_FAST=-O0 but still compiling the run-time library with higher optimization as it can significantly improve tracing speed which you want during a debug cycle. I am concios about adding more complexity to the build so I will do some measurements on this to confirm this indeed makes sense. If it is, adding an OPT_GLOBAL or similar to apply to the run-time library is probably in order. I would think this should default with `OPT_GLOBAL = $(OPT_FAST)`, but overrideable by the user. Let me know if you have anything against this. In the end this can always be achieved with external makery.

If you think we should do this it would be good to get it into the next release in order to break people's makefiles only once (which is not what I am trying to do)!

wsnyder, 2020-05-26:
Concept seems fine.  GLOBAL suggests everything as opposed to the library, but that naming mistake was mine long ago so probably best to use it.

gezalore, 2020-05-26:
I don't like the name GLOBAL much either, so I would be very happy to use something else, 4 letters would be neat.

wsnyder, 2020-05-26:
API?  SHARED?

wsnyder, 2020-05-26:
SHARED suggests shared object, so perhaps not the best choice.

gezalore, 2020-05-26:
AUX / VAUX / MISC / COMMON / COMM / VLRT (Verilator Run Time) ?

gezalore, 2020-05-26:
or VMRT for Verilated Model Run-Time (I do struggle a bit with the difference between VL* VM* and VK* prefixes)

wsnyder, 2020-05-26:
Suggest we keep GLOBAL, we can claim it's for compatibility.

VM_* are generated by Verilator's makefile generator or set by users.  VK_ are set in the makefile itself and intended to be locals for the most part.  Of course nothing checks that so might be some mislabelings.  So, basically you should feel fine changing a VK but a VM might cause a compatibility problem.

wsnyder, 2020-05-26:
(Also suspect we'll be in there again when we split up the global-global stuff versus global-per-version as discussed elsewhere and we'll need more names.)

gezalore, 2020-05-26:
Ok, I'm happy to not think about naming until the reconciliation of globals (there it is again...).

Just to confirm this is useful with some hard numbers:

Warning: way too much data follows.

I noted important points below the tables.
@wsnyred I think you already gone through a lot of this for your recent paper,
but the tracing bit should contain new information as it now includes the
recent improvements.

All tables are EH1 cmark_iccm , OPT_SLOW=-O0, columns are OPT_GLOBAL,
rows are OPT_FAST.

# Clang 10.0

## Build time (wall clock -j8)

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |  -Oz  |
|-----|-------|-------|-------|-------|-------|-------|
| -O0 | 12.30 | 12.75 | 12.87 | 12.94 | 12.74 | 12.62 |
| -O1 | 15.19 | 15.82 | 15.86 | 15.86 | 15.85 | 15.79 |
| -O2 | 25.96 | 26.13 | 26.69 | 26.74 | 25.62 | 26.22 |
| -O3 | 26.09 | 26.02 | 26.36 | 26.73 | 26.32 | 26.30 |
| -Os | 24.97 | 25.33 | 25.55 | 25.17 | 25.36 | 25.32 |
| -Oz | 16.66 | 17.06 | 17.29 | 17.29 | 16.94 | 17.01 |

As expected OPT_GLOBAL makes insignificant difference in build time.
Note `-Oz` for OPT_FAST is an interesting option as it's much faster than
`-Os`.

## Execution time, no tracing

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |  -Oz  |
|-----|-------|-------|-------|-------|-------|-------|
| -O0 | 17.56 | 17.32 | 16.99 | 17.00 | 17.01 | 17.04 |
| -O1 |  8.18 |  7.97 |  7.66 |  7.69 |  7.68 |  7.81 |
| -O2 |  7.26 |  6.91 |  6.64 |  6.67 |  6.79 |  6.81 |
| -O3 |  7.18 |  6.92 |  6.64 |  6.65 |  6.71 |  6.78 |
| -Os |  7.47 |  7.20 |  6.93 |  6.96 |  6.92 |  7.01 |
| -Oz |  7.52 |  7.26 |  6.97 |  7.01 |  7.05 |  7.08 |

Again as expected OPT_GLOBAL makes little difference, but higher
optimization does improve speed a little, `-Os` is as good as any.
Obviously OPT_FAST has a lot of impact, `-O2` `-O3` `-Os` and `-Oz`
are largely the same

## Execution time, VCD tracing

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |  -Oz  |
|-----|-------|-------|-------|-------|-------|-------|
| -O0 | 52.56 | 41.93 | 39.98 | 39.99 | 40.05 | 40.83 |
| -O1 | 34.97 | 25.02 | 23.26 | 23.22 | 23.21 | 24.16 |
| -O2 | 30.49 | 20.43 | 18.65 | 18.67 | 18.78 | 19.63 |
| -O3 | 30.32 | 20.23 | 18.53 | 18.50 | 18.65 | 19.53 |
| -Os | 30.85 | 21.00 | 19.24 | 19.27 | 19.21 | 20.06 |
| -Oz | 34.41 | 24.35 | 22.64 | 22.61 | 22.58 | 23.54 |

This is where OPT_GLOBAL maters, and yields a 30%+ speed improvement at
virtually no noticeable build time cost.

## Execution time, FST tracing

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |  -Oz  |
|-----|-------|-------|-------|-------|-------|-------|
| -O0 | 93.04 | 58.83 | 54.40 | 53.90 | 55.19 | 56.23 |
| -O1 | 74.36 | 40.98 | 36.22 | 36.28 | 37.33 | 38.77 |
| -O2 | 69.61 | 36.68 | 33.05 | 33.08 | 33.88 | 34.82 |
| -O3 | 69.83 | 36.50 | 32.96 | 32.83 | 33.72 | 34.45 |
| -Os | 70.01 | 37.18 | 33.61 | 33.56 | 34.27 | 35.07 |
| -Oz | 73.81 | 40.51 | 35.96 | 36.11 | 37.00 | 38.09 |

Same as for VCD, except the benefit of OPT_GLOBAL is 70%+.

# GCC 9.3

Note GCC has no `-Oz`

## Build time (wall clock -j8)

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |
|-----|-------|-------|-------|-------|-------|
| -O0 | 16.96 | 17.22 | 17.70 | 17.94 | 17.18 |
| -O1 | 21.35 | 21.76 | 21.97 | 22.03 | 21.67 |
| -O2 | 78.73 | 78.45 | 78.97 | 79.33 | 78.82 |
| -O3 | 80.10 | 80.22 | 79.91 | 79.53 | 80.00 |
| -Os | 24.72 | 25.07 | 24.82 | 25.41 | 25.12 |

OPT_GLOBAL has little effect, `-Os` is clear winner for OPT_FAST.

## Execution time, no tracing

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |
|-----|-------|-------|-------|-------|-------|
| -O0 | 16.46 | 16.08 | 16.05 | 16.07 | 16.21 |
| -O1 |  9.17 |  8.98 |  8.97 |  9.01 |  9.01 |
| -O2 |  8.74 |  8.66 |  8.54 |  8.57 |  8.59 |
| -O3 |  8.70 |  8.59 |  8.49 |  8.43 |  8.56 |
| -Os |  8.12 |  7.80 |  7.68 |  7.83 |  7.84 |

OPT_GLOBAL has little effect, `-Os` is clear winner.

## Execution time, VCD tracing

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |
|-----|-------|-------|-------|-------|-------|
| -O0 | 52.61 | 37.74 | 37.64 | 37.29 | 38.46 |
| -O1 | 36.16 | 22.46 | 22.33 | 22.45 | 22.83 |
| -O2 | 36.05 | 22.35 | 22.09 | 21.99 | 22.69 |
| -O3 | 35.89 | 22.26 | 22.05 | 21.96 | 22.69 |
| -Os | 38.52 | 24.49 | 24.28 | 24.31 | 24.98 |

OPT_GLOBAL gets you 40%+ with `-Os`. OPT_FAST actually is worse with `-Os` by
up to 14%. Seems like GCC doesn't like the tracing routines much with `-Os`.

## Execution time, FST tracing

|     |  -O0  |  -O1  |  -O2  |  -O3  |  -Os  |
|-----|-------|-------|-------|-------|-------|
| -O0 | 91.40 | 52.65 | 52.42 | 52.05 | 54.00 |
| -O1 | 76.09 | 37.77 | 37.59 | 37.35 | 38.75 |
| -O2 | 76.59 | 37.73 | 37.37 | 37.23 | 38.77 |
| -O3 | 76.22 | 37.39 | 37.15 | 38.32 | 38.55 |
| -Os | 78.01 | 39.64 | 39.45 | 38.96 | 40.90 |

Same as VCD but OPT_GLOBAL gets you 75%+.

# Remarks

In conclusion -Os indeed seems like the sweet spot for both performance and
build time, bar the generated tracing functions with GCC. Clang's `-Oz` gets
you close to the same performance as `-Os` at much faster build times bar
the tracing routines again. I seem to remember Clang docs saying `-Oz` is
actually what is closer to GCC's `-Os`.

Based on this I will PR OPT_GLOBAL=-Os always, with a recommendation that
the user should not normally need to change this.

wsnyder, 2020-05-26:
I agree with your conclusions, and data seems about what I would expect, so feel free to commit appropriately, thanks.

gezalore, 2020-05-27:
Implemented in #2373

closed 2020-05-27 by gezalore