verilator/verilator #2372: Doc wrongly says "Ranges must be big-bit-endian"

alanmimms, 2020-05-26:
I'm reading the PDF file for Verilator 4.035. The section title is "Ranges must be big-bit-endian" but what it appears to mean is the opposite. If Verilator supports MSB >= LSB in ranges, then it is _little bit-endian_. That is, the LSBs are indexed first in order.

I have a design that is _full_ of ranges like `bit foo[0:35]`, so I was disappointed to read this on the second to last page of the documentation. I can't use Verilator, apparently.

:sob:

gezalore, 2020-05-26:
I will defer to @wsnyder regarding the nomenclature.

Actually, it what you have is `bit foo [0:35]`,  then that is an unpacked array which should be supported just fine. What you can't do is `bit [0:35] foo` which is a packed array.

wsnyder, 2020-05-26:
Thanks for checking on this.  That comment is old for a while now, ranges should be allowed both directions now.  By chance that documentation section was rereviewed a few weeks ago and is already fixed in the git master version.

closed 2020-05-26 by wsnyder

alanmimms, 2020-05-26:
> I will defer to @wsnyder regarding the nomenclature.
> 
> Actually, if what you have is `bit foo [0:35]`, then that is an unpacked array which should be supported just fine. What you can't do is `bit [0:35] foo` which is a packed array.

Wow. My C programming background haunts me _again_. I meant to write `bit [0:35] foo`, which is what I have mostly. And according to @wsnyder reply to this this is now supported!
:grinning:

gezalore, 2020-05-26:
Indeed, apparently I learned something too, thanks :)

alanmimms, 2020-05-26:
Hey, I just tried my source base against the git master checked out today. I still am getting warnings about bit-endianness:

```
   20 |   bit [0:17] ARL;
      |       ^
%Warning-LITENDIAN: edp.sv:21:7: Little bit endian vector: MSB < LSB of bit range: 0:17
```
Did I misunderstand your point @wsnyder? It appears MSB < LSB is not actually supported? If so I'm :cry:.

wsnyder, 2020-05-26:
It's a style warning. See the manual.  "Ignoring this warning will only suppress the lint check, it will simulate correctly.

alanmimms, 2020-05-27:
I got much further. I think I wasn't understanding completely what it meant. I disabled that particular warning and I'm making good progress. Verilator is finding a _lot_ of lint in my code and most of the issues are actually things I need to fix. So even if I don't succeed in creating an executable sim model from this exercise it will be really useful to have done it. But I think I can get an executable sim and maybe even make really good progress without ever launching Vivado again until I'm ready to move to hardware. That would make me happy.