verilator/verilator #2385: --trace-max-width is ignored

gezalore, 2020-05-30:
This option is documeted and accepted but has been ignored since Dec 2013:
00724597f447e6ebab14b950b30280286750350

What would be prefered:
- Remove (given that nobody complained in 6 years?)
- Warn deprecated, remove in 12 months
- Fix

BTW there are other options that are ignored too like `--trace-dups` which are accepted, still have accessors in the options, are undocumented and are unused internally.

wsnyder, 2020-05-30:
I'd suggest some mechanism this or otherwise to prevent tracing anything that results in data that is too expensive to detect changes on, as this will bite someone at some point.  An alternative would be to look at widthTotalBytes.

traceDups can be removed. Any others?

gezalore, 2020-05-30:
There are some more unused function in V3Option but I think those 2 are actually the only unused options. Will nuke traceDdups. I will fix the maxWidth one at a later point.

wsnyder, 2020-05-30:
BTW coverage unfortunately doesn't find this (https://codecov.io/gh/verilator/verilator/src/master/src/V3Options.cpp) as the conditions don't affect line coverage.  Maybe time to clang-reformat that then they will each get their own line.

gezalore, 2020-05-30:
Yeah that's exactly what it made me think about our `if (cond) single` code style. Without branch coverage it is misleading.

wsnyder, 2020-05-30:
I can experiment with gcov branch analysis.

Verilator code coverage IMO (of course) is much better than GCC's as it can do variable toggle coverage which would make this quite obvious.  Also much nicer disables.  Anyhow.

gezalore, 2020-05-30:
Maybe time for the C++ frontend? :wink:

wsnyder, 2020-06-29:
This was resolved earlier, thanks @gezalore.

Also as to the side conversation, Verilator internal coverage now shows branch coverage - so we can now see holes like this.

closed 2020-06-29 by wsnyder