verilator/verilator #2390: Add FreeBSD build to Travis CI

gezalore, 2020-05-31:
Travis now has FreeBSD support (https://github.com/travis-ci/travis-yml/pull/151/files). We would like chron builds to include FreeBSD to maintain compatibility. Ideally we would welcome someone with FreeBSD experience to maintain this.

mjkillough, 2020-05-31:
I don't have much FreeBSD experience, but I've started a branch to try and get this working. It's currently compiling and passing most tests, but there are a few tests core dumping which I need to investigate.

(My [very rough WIP is here](https://github.com/verilator/verilator/compare/master...mjkillough:master), but it disables the current tests and runs FreeBSD as `!= cron` in order to make it easier to test).

mjkillough, 2020-05-31:
I forgot to mention, caching in Travis CI appears to be broken for FreeBSD. This means the tests will run very slowly (as they need to re-compile for each step) if I do get them running. I'm hoping travis-ci/casher#54 will allow us to use `ccache`.

wsnyder, 2020-05-31:
Thanks. If you can just get it to build and run "make examples" that would prevent 95% of the problems.

Without ccache I don't think you'll be able to run all of the tests, as you will run out of CPU time (Travis limits to ~50 minutes).

mjkillough, 2020-06-01:
I've opened a PR to just add `make examples` for now. I'm happy to come back and get the tests running once the `ccache` is fixed, but I agree they'll be too slow at the moment.

closed 2020-06-30 by gezalore