verilator/verilator #2391: Segmentation fault(core dumped) during make verilate in Ariane project

NazerkeT, 2020-05-31:
Dear Verilator Team, I am working on an Ariane project and using a verilator tool. However, for now, I am struggling with `make verilate`. Following is the description of the problem.

I have installed riscv-gnu-toolchain, riscv-tool with the latest versions available, updated Ariane submodules recursively, installed Verilator with the latest (4.035) version globally from the git master branch, unset $VERILATOR_ROOT by every built, have passed Verilator test examples successfully, downloaded some packages that advised by Verilator page that can help to improve performance of the builds, and so far it seems that I have done everything right following manuals of both [Ariane](https://github.com/pulp-platform/ariane) and [Verilator](https://github.com/verilator/verilator). Also, I am not using SystemC, so I did not download that package, if it may be the case. However, for my Ubuntu 18.04 OS, I can't `make verilate` with `Segmentation fault (core dumped) %Error: Command Failed` error. There are plenty of warnings on pin assignments, delays and expected values which you may see in the attached terminal log file.

Could you help me with the solution of this problem? Initially, I thought that it could be because of my RAM size (6GB) being small, however, it seems it is not the case.
I have googled around old issues, but for now, I did not find anything useful.
Hope to hear back from you soon.

[make_verilate_log.txt](https://github.com/verilator/verilator/files/4708136/make_verilate_log.txt)

wsnyder, 2020-05-31:
Sorry you are having problems.

Are you sure it's not out of memory?  Typically unless out of memory the wrapper would catch any seg fault and print a different message. Also you may want to run with "top" in another shell and see if you are running out.

If you run with "--debug --gdbbt" GDB should show where it is stopping, post those results please.

mikemusasa, 2020-06-02:
Hi, I am having the same issue. 
do you mean to  run :  make verilate -- debug --gdbbt ?  or how can one run the debug-- gdbbt so that it shows where it is stopping ?

wsnyder, 2020-06-02:
Whatever runs verilator (e.g. the makefile) needs to add to the verilator command line "--debug --gdbbt".

gezalore, 2020-06-03:
```
Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7c74859 in __GI_abort () at abort.c:79
#2  0x000055555595b26c in V3Error::vlAbort () at ../V3Error.cpp:162
#3  0x000055555595bc70 in V3Error::v3errorEnd (sstr=..., locationStr=...) at ../V3Error.cpp:275
#4  0x00005555559757de in FileLine::v3errorEnd (this=0x555557c89cc0, sstr=..., locationStr=...) at ../V3FileLine.cpp:374
#5  0x000055555576d7d5 in AstNode::v3errorEnd (this=0x55555d298e40, str=...) at ../V3Ast.cpp:1202
#6  0x000055555576d0d5 in AstNode::v3errorEndFatal (this=0x55555d298e40, str=...) at ../V3Ast.cpp:1145
#7  0x00005555557b14c4 in BrokenCheckVisitor::processAndIterate (this=0x7fffffffa770, nodep=0x55555d298e40) at ../V3Broken.cpp:253
#8  0x00005555557b197b in BrokenCheckVisitor::visit (this=0x7fffffffa770, nodep=0x55555d298e40) at ../V3Broken.cpp:275
#9  0x00005555557436a8 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d298e40) at ./V3Ast__gen_visitor.h:16
#10 0x000055555574a16c in AstAssign::accept (this=0x55555d298e40, v=...) at ../V3AstNodes.h:3162
#11 0x000055555576abc9 in AstNode::iterateAndNextConst (this=0x55555d298e40, v=...) at ../V3Ast.cpp:874
#12 0x000055555576a7a8 in AstNode::iterateChildrenConst (this=0x55555d287ce0, v=...) at ../V3Ast.cpp:807
#13 0x00005555557affe5 in AstNVisitor::iterateChildrenConst (this=0x7fffffffa770, nodep=0x55555d287ce0) at ../V3Ast.h:2876
#14 0x00005555557b189d in BrokenCheckVisitor::processAndIterate (this=0x7fffffffa770, nodep=0x55555d287ce0) at ../V3Broken.cpp:271
#15 0x00005555557b1aff in BrokenCheckVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287ce0) at ../V3Broken.cpp:283
#16 0x0000555555746070 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287ce0) at ./V3Ast__gen_visitor.h:208
#17 0x0000555555743a28 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287ce0) at ./V3Ast__gen_visitor.h:32
#18 0x000055555578f324 in AstBegin::accept (this=0x55555d287ce0, v=...) at ../V3AstNodes.h:4497
#19 0x000055555576abc9 in AstNode::iterateAndNextConst (this=0x55555d287ce0, v=...) at ../V3Ast.cpp:874
#20 0x000055555576a7f0 in AstNode::iterateChildrenConst (this=0x55555d298cc0, v=...) at ../V3Ast.cpp:809
#21 0x00005555557affe5 in AstNVisitor::iterateChildrenConst (this=0x7fffffffa770, nodep=0x55555d298cc0) at ../V3Ast.h:2876
#22 0x00005555557b189d in BrokenCheckVisitor::processAndIterate (this=0x7fffffffa770, nodep=0x55555d298cc0) at ../V3Broken.cpp:271
#23 0x00005555557b1aff in BrokenCheckVisitor::visit (this=0x7fffffffa770, nodep=0x55555d298cc0) at ../V3Broken.cpp:283
#24 0x0000555555746508 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d298cc0) at ./V3Ast__gen_visitor.h:229
#25 0x00005555557483a8 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d298cc0) at ./V3Ast__gen_visitor.h:369
#26 0x000055555578e99e in AstWhile::accept (this=0x55555d298cc0, v=...) at ../V3AstNodes.h:4239
#27 0x000055555576abc9 in AstNode::iterateAndNextConst (this=0x55555d284a80, v=...) at ../V3Ast.cpp:874
#28 0x000055555576a7a8 in AstNode::iterateChildrenConst (this=0x55555d287b20, v=...) at ../V3Ast.cpp:807
#29 0x00005555557affe5 in AstNVisitor::iterateChildrenConst (this=0x7fffffffa770, nodep=0x55555d287b20) at ../V3Ast.h:2876
#30 0x00005555557b189d in BrokenCheckVisitor::processAndIterate (this=0x7fffffffa770, nodep=0x55555d287b20) at ../V3Broken.cpp:271
#31 0x00005555557b1aff in BrokenCheckVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287b20) at ../V3Broken.cpp:283
#32 0x0000555555746070 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287b20) at ./V3Ast__gen_visitor.h:208
#33 0x0000555555743a28 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287b20) at ./V3Ast__gen_visitor.h:32
#34 0x000055555578f324 in AstBegin::accept (this=0x55555d287b20, v=...) at ../V3AstNodes.h:4497
#35 0x000055555576abc9 in AstNode::iterateAndNextConst (this=0x55555d287b20, v=...) at ../V3Ast.cpp:874
#36 0x000055555576a7a8 in AstNode::iterateChildrenConst (this=0x55555d287a40, v=...) at ../V3Ast.cpp:807
#37 0x00005555557affe5 in AstNVisitor::iterateChildrenConst (this=0x7fffffffa770, nodep=0x55555d287a40) at ../V3Ast.h:2876
#38 0x00005555557b189d in BrokenCheckVisitor::processAndIterate (this=0x7fffffffa770, nodep=0x55555d287a40) at ../V3Broken.cpp:271
#39 0x00005555557b1aff in BrokenCheckVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287a40) at ../V3Broken.cpp:283
#40 0x0000555555746070 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287a40) at ./V3Ast__gen_visitor.h:208
#41 0x0000555555743a28 in AstNVisitor::visit (this=0x7fffffffa770, nodep=0x55555d287a40) at ./V3Ast__gen_visitor.h:32
#42 0x000055555578f324 in AstBegin::accept (this=0x55555d287a40, v=...) at ../V3AstNodes.h:4497
#43 0x000055555576abc9 in AstNode::iterateAndNextConst (this=0x55555d287a40, v=...) at ../V3Ast.cpp:874
#44 0x000055555576a7cc in AstNode::iterateChildrenConst (this=0x55555d298a80, v=...) at ../V3Ast.cpp:808
#45 0x00005555557affe5 in AstNVisitor::iterateChildrenConst (this=0x7fffffffa770, nodep=0x55555d298a80) at ../V3Ast.h:2876
#46 0x00005555557b189d in BrokenCheckVisitor::processAndIterate (this=0x7fffffffa770, nodep=0x55555d298a80) at ../V3Broken.cpp:271
[... cut ...]
#90 0x000055555573a724 in verilate (argString=...) at ../Verilator.cpp:547
#91 0x000055555573b849 in main (argc=225, argv=0x7fffffffae98, env=0x7fffffffb5a8) at ../Verilator.cpp:643
A debugging session is active.
```

There are a lot of warnings/errors prior.

To reproduce:
```
git clone https://github.com/pulp-platform/ariane.git
cd ariane
git submodule update --init --recursive
export RISCV=none
make verilator=/home/gl/work/verilator/src/bin/verilator verilate
```

I will probably have little time to investigate this prior to the release.

gezalore, 2020-06-03:
And what I meant to say is that I will likely not have time to investigate this prior to the release.

mikemusasa, 2020-06-03:
@gezalore  is a new version of the verilator planned to be released ? 

this is the error i am getting : 

%Error: Cannot find file containing module: tb/dpi/msim_helper.cc--debug
        ... Looked in:
             src/axi_node/tb/dpi/msim_helper.cc--debug
             src/axi_node/tb/dpi/msim_helper.cc--debug.v
             src/axi_node/tb/dpi/msim_helper.cc--debug.sv
             src/common_cells/include//tb/dpi/msim_helper.cc--debug
             src/common_cells/include//tb/dpi/msim_helper.cc--debug.v
             src/common_cells/include//tb/dpi/msim_helper.cc--debug.sv
             tb/dpi/msim_helper.cc--debug
             tb/dpi/msim_helper.cc--debug.v
             tb/dpi/msim_helper.cc--debug.sv
             work-ver/tb/dpi/msim_helper.cc--debug
             work-ver/tb/dpi/msim_helper.cc--debug.v
             work-ver/tb/dpi/msim_helper.cc--debug.sv

gezalore, 2020-06-03:
Yes there will be a new release soon. You are missing a space before "--debug" on your command line.

mikemusasa, 2020-06-03:
@gezalore  thank you for the quick answer. 
After doing that on the $(verilate_command)   line in the make file, line 403 , which now looks like this : $(verilate_command) --debug --gdbbt 
 I am now  back to seeing the original error message  : 

Makefile:403: recipe for target 'verilate' failed
make: *** [verilate] Error 134 

Is the new release happening soon or in a couple of months ?

wsnyder, 2020-06-03:
The problem was related to the merge to fix #2223, this is fixed.  However when I compile and have warnings turned on I get:

```
Variane_testharness__8.cpp: In static member function 'static void Variane_testharness::_combo__TOP__32(Variane_testharness__Syms*)':
Variane_testharness__8.cpp:11691:147: error: expected ';' before '}' token
11691 |             vlTOPp->ariane_testharness__DOT__i_ariane__DOT__csr_regfile_i__DOT__csr_update__DOT__unnamedblk1__DOT__unnamedblk1__DOT____Vincrement1}
      |                                                                                                                                                   ^
      |                                                                                                                                                   ;
Variane_testharness__8.cpp:11691:21: error: statement has no effect [-Werror=unused-value]
```

This is also coming through from the #2223 changes.  @tgorochowik can you please take a look & also get a snippit from whatever is causing this into test_regress?  Thanks.

closed 2020-06-03 by wsnyder

NazerkeT, 2020-06-04:
> Please try the new master with @tgorochowik fix, seems OK to me.

Thanks for quick collaboration! Revisiting issue 2400 and last comments, does it mean that now I can git pull the Verilator and continue with testing? Or should I necessarily wait for a new release?

wsnyder, 2020-06-04:
>Starting Verilator 4.035 devel rev v4.034-173-gb60a92ee

This doesn't seem the most recent, can you try pulling and recompiling again?

NazerkeT, 2020-06-04:
I have tried several times, but except first pull result is same: `Already up to date.`
Here are the terminal outputs:

`(base) nazerke@nazerke-Lenovo-G50-70:~$ cd verilator
(base) nazerke@nazerke-Lenovo-G50-70:~/verilator$ verilator --version
Verilator 4.035 devel rev v4.034-173-gb60a92ee
(base) nazerke@nazerke-Lenovo-G50-70:~/verilator$ git pull
Already up to date.
(base) nazerke@nazerke-Lenovo-G50-70:~/verilator$ git log --graph
* commit 79f228b04f0f95e72ae68068b27412f8433ebc1b (HEAD -> master, origin/master, origin/HEAD)
| Author: Wilson Snyder <wsnyder@wsnyder.org>
| Date:   Wed Jun 3 20:11:09 2020 -0400
| 
|     Remove dead code (bisonreader)
| 
* commit d7cc1e6844b28ed9ded101cc00d5a94221af29b2
| Author: Wilson Snyder <wsnyder@wsnyder.org>
| Date:   Wed Jun 3 20:05:13 2020 -0400
| 
|     Tests: Use lint in many negative tests to avoid running under both vlt and vltmt
| 
* commit 42312101b5d253988a87a4cdec9962d3db3f84e9
| Author: Wilson Snyder <wsnyder@wsnyder.org>
| Date:   Wed Jun 3 19:36:50 2020 -0400
| 
|     Tests: Add array OOB tests as unsupported (#792)
| 
* commit 329f5b712aa9ea1ad4232459aa24345e6b470b8f
| Author: Wilson Snyder <wsnyder@wsnyder.org>
| Date:   Wed Jun 3 19:19:39 2020 -0400
:
`

NazerkeT, 2020-06-04:
> Then it should say
> Verilator 4.035 devel rev v4.034-194-g79f228b04
> 
> if not most likely you aren't reompililg it, aren't installing it, or you are running a version installed from somewhere else.

Could you please look at git log? I have updated my comment. Nevertheless, I have loaded verilator from this repository, I will check it again.

NazerkeT, 2020-06-05:
Hi, there. Today, I have deleted my verilator repo, downloaded it once again from this git and run `make verilate` with success. Thank you very much!

It turned out that, git repo mentioned under 2.3. Git Installation guideline in [Veripool](https://www.veripool.org/projects/verilator/wiki/Installing) website is somewhat different from the current git repo. As far as I have understood this could be reason why I was not able to follow the last merge. I may be mistaken too. Hope this comment will help others.

mikemusasa, 2020-06-08:
Hi, there.  I deleted my verilator repo , downloaded a new one and checked the version that it was the latest. While trying make verilate again , it went further than in my earlier trials  but failed again. 
Don't know if something happened since the last fix. 
Here is the error log i copied and saved :

wsnyder, 2020-06-08:
>as: unrecognized option '--64'
Something is wrong in the make or compiler flags - see where "-64" are in them.  This isn't (likely) related to Verilator.

PR #2400 (2020-06-03): Fix Ariane issues with PostAdd nodes
Fixes #2391

For now this just adds a minimal code snippet that reproduces it. I will try to submit a fix when I find it.