verilator/verilator #2397: assignment did not succeed in always_ff when clk_gated driven

hdzhangdoc, 2020-06-03:
Hello，assignment did not succeed in code below when translation to systemc

`258    always_ff @(posedge clk_gated, negedge rst_n)
259    begin
260       if(rst_n == 1'b0)
261       begin
262       for (i=0; i< DATA_DEPTH; i++)
263          FIFO_REGISTERS[i] <= {DATA_WIDTH {1'b0}};
264       end
265       else
266       begin
267          if((grant_o == 1'b1) && (valid_i == 1'b1))
268             FIFO_REGISTERS[Push_Pointer_CS] <= data_i;
269       end
270    end`

[image]

it it found under master version

wsnyder, 2020-06-03:
You don't show the logic creating clk_gated, but given the name "clk_gated" presumably it is a gated clock.  Please see the manual about needing to mark clock enables, specifically this: 

   reg enable_r /*verilator clock_enable*/;
   wire gated_clk = clk & enable_r;

If that does not help please post a stand-alone example and command line that can make the waveforms you show, thanks.

hdzhangdoc, 2020-06-04:
> You don't show the logic creating clk_gated, but given the name "clk_gated" presumably it is a gated clock. Please see the manual about needing to mark clock enables, specifically this:
>
> ```
>    reg [...]

I tried this method but it didn't work.
I also extract the module out into sc_examples project, and simulate all input as same with above, it works as wanted. 
Now I don't know why.

hdzhangdoc, 2020-06-04:
I read the generated systemC code and found that this “clk_gated“ signal is not used as the clock signal to drive FIFO_REGSIGER , and FIFO_REGISTER has no reassignment code，I suspect that the code generation of this piece is affected by some kind of error

wsnyder, 2020-06-04:
Look at where you are setting inputs versus clocks relative to the eval calls, probably there is a race.

wsnyder, 2020-06-04:
>Look at where you are setting inputs versus clocks relative to the eval calls, probably there is a race.

This was in regards to systemc working and c++ not working.

I don't understand your other comment about no code around FIFO_REGISTER as if that was the case your waves would not show the signal moving.

hdzhangdoc, 2020-06-04:
This is the project I built after extracting this module, but unfortunately the waveform of this small test case is consistent with vcs, I don’t know why the above problem appeared in the original large project

wsnyder, 2020-06-28:
I think we're waiting for a test case on this. If you think otherwise please let us know, or if you are all set as-is please indicate we can close it.

wsnyder, 2020-08-10:
If you have a test case please reopen, otherwise closing.

closed 2020-08-10 by wsnyder