verilator/verilator #2402: protect-lib for SV Pkg support

fuyong10, 2020-06-04:
I am playing with Ariane from https://github.com/pulp-platform/ariane

1. Verilator compiled and run nicely
2. To try protect-lib, I select a submodule "frontend". Compile done with no error.
3. Simple add "-protect-lib FE_secret":
      %Error: src/frontend/frontend.sv:31: Internal: Unknown node type reached emitter: REFDTYPE 'bp_resolve_t'
      input  bp_resolve_t        resolved_branch_i,
                  ^~~~~~~~~~~~
   The generated file Vfrontend.cpp, Vfrontend.h are exactly the same as generated in #2. 
   But the generated FE_secret.sv contains:
   
     input 
     ???? // DEFDTYPE 'bp_resolve_t'
       esovled_branch_i
   
    seems like struct bp_resolve_t is not recognized, or the definition could not be found.

bp_resolve_t is defined in package file "ariane_pkg.sv" as: 
    typedef struct packed {
        logic                   valid;           // prediction with all its values is valid
        logic [riscv::VLEN-1:0] pc;              // PC of predict or mis-predict
        logic [riscv::VLEN-1:0] target_address;  // target address at which to jump, or not
        logic                   is_mispredict;   // set if this was a mis-predict
        logic                   is_taken;        // branch is taken
        cf_t                    cf_type;         // Type of control flow change
    } bp_resolve_t;

Verilator Verison:
    Verilator 4.029 devel rev v4.028-4-g18f8cd05

Will try to create a smaller test case but I assume this might be something already know ?

toddstrader, 2020-06-04:
Only native types have been tested so far, so it's not known but also not a great surprise.

A test in the form of `test_regress/t_prot_lib.*` would be helpful.  Also, would you be interested in working on a patch?

fuyong10, 2020-06-04:
thanks Tod. So since you already marked the data type as "????", I assume maybe an work around is to simple figure out the possible data type and add that in here? 

I would be very interested in working on a patch but just start to read the documentation. Just not sure how much knowledge is needed for this and need some guide for sure.

toddstrader, 2020-06-04:
Agreed that determining the data type is the key here.  We'll need to be able to find the fully scoped type name `ariane_pkg::bp_resolve_t` in this case.  Additionally, we'll need to determine how to pass this through the DPI, but since it's a packed type it will be easier since it's essentially the same thing as a packed vector.

It sounds like you're already working on a smaller test.  If you can make that in the form of `test_regress` tests, then we'll have something for the regression suite.  And if you're interested in hacking on the C++ I can give you some pointers in that direction once we have the test.

fuyong10, 2020-06-04:
the test is done with Ariane RISCV core. Not a big design in terms of the design size but there are quite some coding. I can trim down a test case which can be put into test_regress. Pls give me the instruction how to make it into regress then as the first step.

fuyong10, 2020-06-04:
Also maybe this could be a PR. Any chance to output an instance (and sub-instances under it) as a protected_ip? Now protect-lib can only output for the top module. 

I also tried to compile the full Ariance but want to output the instance called "frontend" only by changing the "-top-module" from "ariane" to "frontend". Compiler reports error says "frontend" is not top as it is instantiated by "ariane".  If veriltor can allow full design varilated but just output the instance, will the data type already revolved?

toddstrader, 2020-06-05:
Instructions for making a new test can be found here:
https://github.com/verilator/verilator/blob/master/docs/internals.adoc#testing

Regarding the --top-module question, if you're thinking about incremental compilation using --protect-lib, then there are a number of threads discussing this already.  I'd suggest starting with #2206 .

If you're just wondering how to trim down the Ariane core to reproduce the problem, I'd suggest now that we understand what the problem is you should be able to create a test to reproduce it without using any of the Ariane code.  I believe all you need to do is create a user defined type (could be a packed struct since that's your current problem) and pass it through a sub-module that you're going to --protect-lib.  `t_prot_lib` should give you a template for how to do this.

fuyong10, 2020-06-06:
While I will continue learn to generate new test. The test case has been attached here. "runme" is the command.
[t.zip](https://github.com/verilator/verilator/files/4739395/t.zip)

margej, 2020-11-05:
I think #2611 can be closed as a duplicate of this issue (although it might help prevent future dups to rename this issue to include the error string)

@toddstrader, if --protect-lib could just convert any packed struct I/Os to the equivilent multi-bit wire port, that would be sufficient for the instancing module to connect that port _directly_ to a struct of the same type (I have confirmed this works with VCS and Xcelium)

wsnyder, 2026-10-18:
Have been many improvements in this over the years.  If there's specific requirements needed here please file a new issue describing the specific sub-feature needed, thanks.

closed 2026-10-18 by wsnyder