verilator/verilator #2404: How to speed up compile time for "top_ALLslow.o"

ayewinoung, 2020-06-04:
How may we help - what is your question?
Hi - I've recently upgraded Verilator to ver 4.034, and I've observed improvements over older version I was using, esp on multithreading and thank you for that.

But I've observed significant increased in compile time, I'm *not* sure whether its because of new version of tool or whether I start compiling for multithreading with "--threads" with switch "-std=c++14" to be compatible with SystemC compile.

In particular it spent a lot of time on compiling "top_ALLfast.o and top_ALLslow.o":

Is there anything I'm doing it wrong ? any advice on settings ? Any pointers appreciated.

gezalore, 2020-06-04:
Glad to hear you are seeing improvements.
Regarding build time, the current master branch includes changes aimed at imporving C++ compilation times, so it would be worth trying that first.
For earlier releases, for medium or larger designs, you should probably use VM_PARALLEL_BUILDS=1 when invoking `make`. Please see the section of the manual describing this variable.

ayewinoung, 2020-06-04:
ok thanks, will try it out and report back.. feel free to close the thread

closed 2020-06-04 by gezalore

gezalore, 2020-06-04:
Great. Probably should also recommend reading the updated BENCHMARKING & OPTIMIZATON section in bin/verilator and the parts about OPT_FAST and OPT_SLOW. Please let us know of the results, we are interested in the user experience.