verilator/verilator #2406: Internal coverage in V3MergeCond

wsnyder, 2020-06-05:
@gezalore I've been looking at some internal code coverage issues and after cleaning up some false negatives, see only one uncovered in V3MergeCond.cpp:

     160        AstNode* maskLsb(AstNode* nodep) {
     161           if (nodep->width1()) {
     162 --->:         return nodep;
     163           } else {
     164               AstNode* const maskp = new AstConst(nodep->fileline(), AstConst::LogicTrue());
     165               return new AstAnd(nodep->fileline(), nodep, maskp);

I think you suggested this was related to a SweRV test.  Might you be able to check in a test for it?

I'm only picking on this because I think it's fresh in your mind, there's far bigger holes.  Feel free to punt this if you prefer or if it ends up non-trivial to check.

I suppose another alternative is to add verilator_ext_tests to coverage, but I'm reluctant to rely on external designs for any internal testing (as they might change, or be removed when become stale).

gezalore, 2020-06-05:
>I'm only picking on this because I think it's fresh in your mind, there's far bigger holes.

That's totally fair and a good idea so pleast keep doing so. I will see what can be done shortly.

BTW is there an easy way to check coverate of a single test only, so I can see it covered locally before pushing?

On another note, there are still known unreferenced functions (e.g.: V3GraphVertex::inHash()/outHash()), though they seem like generic utlities/algoritms (albeit added years ago and never used). What is the policy for these?

wsnyder, 2020-06-05:
I don't currently have a automated way to run coverage on one test but will figure something out.

I tried to remove any uncovered functions that I felt would never be used, I think the remaining might have eventual use and so would deserve coverage perhaps in a unit-level self test, but of course there's a good amount of work backlog in writing those.

BTW a few minor holes remaining in tracing but nothing I think worthy of attention.

wsnyder, 2020-06-05:
BTW you can easily see if you hit it if you add a UINFO(0, "FIXME HIT IT"); to that branch.  (Use the FIXME so the tests will fail if you commit it.)

wsnyder, 2020-06-05:
Pull, and you can now run e.g.

   nodist/code_coverage --test t_EXAMPLE.pl

then see nodist/obj_dir/coverage/html/index.html in a browser.

Don't run the whole report without the --test flag unless you have a big machine or want a long wait. (~15 minutes with 32 cores).

gezalore, 2020-06-05:
This actually seems like a bug (in MergeCond) `width1()` tests `width() == 1`, but seems like I need `widthMin() == 1`. In particular for this arraysel should already covert the missing line:

https://github.com/verilator/verilator/blob/c5d61da5d2735f2a011e3e4708d706e2217a13d6/test_regress/t/t_merge_cond.v#L70

But what I see is:
```
masking:  ARRAYSEL 0x55555b9ea900 <e707632#> {d70bi} @dt=0x55555c1297a0@(G/wu32/1)
masking: 1: VARREF 0x55555b9e5b00 <e24984#> {d70be} @dt=0x55555859e0d0@(w1)u[63:0]  t__DOT__unnamedblk1__DOT__bits [RV] <- VAR 0x5555586f3980 <e82784#> {d36al} @dt=0x55555859e0d0@(w1)u[63:0]  t__DOT__unnamedblk1__DOT__bits [VSTATIC]  VAR
masking: 2: CONST 0x55555b9e9ce0 <e588552#> {d70bj} @dt=0x55555c1296c0@(G/swu32/6)  6'h1
- V3MergeCond.cpp:162: nodep->widthMin() is 1
- V3MergeCond.cpp:163: nodep->width() is 32
- V3MergeCond.cpp:164: nodep->width1() is 0
```

Do you agree I should use "widthMin() == 1" istead of "width1()" everywhere in V3MergeCond?

gezalore, 2020-06-05:
Hmm, actually that does not quite work either because then `cond & crc[0]` fails as the rhs has a cast with a widthMin() of 1 yet it needs masking:
```
CCAST 0x555558d19d40 <e708739#> {d82bh} @dt=0x5555577a8460@(G/wu32/1) sz32
1: VARREF 0x5555577917a0 <e697090#> {d82be} @dt=0x55555767f5e0@(G/w64)  t__DOT__crc [RV] <- VAR 0x5555577c1680 <e23385#> {d16ap} u1=0x1 @dt=0x55555767f5e0@(G/w64)  t__DOT__crc [VSTATIC]  VAR
```

We do definitely want to avoid the unecessary mask though as the C compiler cannot remove it because it doesn't know the value can only be 0 or 1.

wsnyder, 2020-06-05:
I agree you probably do not want width1. That explains why width1 is never covered as by this stage everything is width() 32/64 etc.

V3Clean looks at both the width of the child and parent to determine where masking is needed.  Replicating that would be a pain, maybe the thing to do is add the And  unless the child has nodep->cleanOut() set?

gezalore, 2020-06-05:
CCast has cleanOut true, so that would not work for the above example. As a precaution I dropped the "if" and just always mask, this should be rare anyway (it's there to allow merging longer runs of ternaries as often there are these reduced forms interspersed) so it should not be a performance issue.

closed 2020-06-05 by gezalore

wsnyder, 2020-06-06:
Seems reasonable, we can probably figure out a post-optimization in e.g. V3Const if it becomes a problem.  Thanks.