verilator/verilator #2407: Error instability in V3SplitVar

wsnyder, 2020-06-05:
@yTakatsukasa

1. We noticed that the t_split_var_1_bad test was occasionally failing e.g.

https://travis-ci.com/github/verilator/verilator/jobs/343524573

The test fails because the error output changes.  I believe this is because in V3SplitVar.cpp's split() there is an unordered_map which can report errors.  

Would you be willing to take a look?   Note we wouldn't want to just make it a std::map as a key of pointers still would not be stable - it needs some stable sort.

2.  I've been trying to look at code coverage, if you look at the coverage report here: https://codecov.io/gh/verilator/verilator/src/master/src/V3SplitVar.cpp

you'll see a bunch of lines that the tests aren't covering. My primary concern would be lines unrelated to errors, but ideally we'd cover the user errors too.  No need to cover internal errors - I tried to turn off coverage on all internal-only errors but might have missed some.

Thanks!

yTakatsukasa, 2020-06-05:
Sure, will do.

wsnyder, 2020-06-05:
Great, appreciated.

To avoid travis failures I temporarily updated test_regress/t/t_split_var_1_bad.pl, please revert that file (see comments there) when you think you have a fix.

yTakatsukasa, 2020-06-05:
Thank you for the temporary fix.

Unfortunately I cannot reproduce the sporadic failure. Maybe memory allocator and ASLR may affect.
So I am testing on [travis ](https://travis-ci.com/github/yTakatsukasa/verilator/builds/169830272) too.
AstNodeComparator has been used to get deterministic message order, but some containers didn't used it.
Now it is used in all containers as https://github.com/yTakatsukasa/verilator/commit/46e4e9b6de023741072eff3aaa71fed7d0a469b9.

If the change seems to stabilize the test, I will submit a PR.

Then I will add tests to increase coverage.

P.S. I am slowly but still working on #2206 . parameterized block is also working on my local copy.

PR #2408 (2020-06-10): Fix V3SplitVar test stability
This PR fixes #2407 .
AstNodeComparator is used in all associative container.

e48bc11 reverts changes of t_split_var_1_bad.{pl,out} in 5a622fd9db43592d996883a1051c87983e2439ee. (driver.pl and t_vlcov_merge.pl are not reverted).
closed 2020-06-10 by wsnyder