verilator/verilator #2411: [Q&A][syntax] AST XML is missing delay assignments

ted-xie, 2020-06-09:
Hi,

Let's say I want to use Verilator not for any real RTL simulation, but just to dump out the AST from some Verilog designs. If I have a code segment like so:

```
reg clk;
always @(*) begin
  clk = 1'b0;
  #1;
  clk = 1'b1;
  #2;
end
```

When I run this through `verilator myfile.v --xml-only`, I can see that Verilator understands the separate assignments of `clk = 1'b0` and `clk = 1'b1`, but not the delay statements. I'm not sure if this is "by design" (since Verilator doesn't support arbitrary unit-delays), or if there is some kind of extra flag that can at least include this syntax in the AST (even if it's effectively a no-op from the code generation's perspective).

ted-xie, 2020-06-09:
I forgot to include my Verilator version.

```
$ verilator --version
Verilator 4.019 devel rev v4.018-12-g900115bb
```

wsnyder, 2020-06-09:
In general, the XML files are only guarenteed to include those constructs which Verilator itself supports.

If you use a more recent version the delays MIGHT be included as they are now parsed.  If not it would be reasonable to move the stripping of them until later (past where XML dumps). In general we're trying to move towards a better separation of parsing/XML dumping and later dropping of unsupported features.

wsnyder, 2020-06-09:
Closing as think is answered, feel free to followup.

closed 2020-06-09 by wsnyder