verilator/verilator #2413: Part select can have incorrect behaviour if LSB is multiplied by a power of 2

crmymh, 2020-06-10:
Consider:

```
   reg [63:0] a;
   reg [2:0]  b;
   wire [7:0] c = a[b * 8 +: 8];
```

Verilator fails with:
`%Warning-WIDTH: t/t_select_plus_mul_pow2.v:16:20: Bit extraction of var[63:0] requires 6 bit index, not 3 bits.`

With -Wno-WIDTH, the result of `c` is always the 8 least-significant bits of `a`, as you'd expect from the warning.

The problem seems to be in the constant folding that happens here:
https://github.com/verilator/verilator/blob/master/src/V3WidthSel.cpp#L462

`b * 8` gets converted to `b << 3`, which according to the Verilog spec should have a resulting size equal to `b` (3 bits).
https://github.com/verilator/verilator/blob/master/src/V3Const.cpp#L881

It works as expected if you comment out the `replaceMulShift` optimization:
https://github.com/verilator/verilator/blob/master/src/V3Const.cpp#L2325

I'm using V4.034. Sample regression test attached:
[t_select_plus_mul_pow2.tar.gz](https://github.com/verilator/verilator/files/4756415/t_select_plus_mul_pow2.tar.gz)

crmymh, 2020-06-11:
Sure, I'll give it a shot. Did you mean `operandSameSize($lhsp, $rhsp)`? I think we need to check that `lhsp->width() == rhsp->width()`?

wsnyder, 2020-06-11:
Fixed with your pull request, thanks for reporting and repairing.

closed 2020-06-11 by wsnyder