verilator/verilator #2420: Upgrading from 4.024 to 4.036 results in more files and hence slower C++ compilation

jsn1993, 2020-06-13:
How may we help - what is your question?
I used the same command to verilated same file using 4.024 and 4.036. However, 4.036 results in more files (*Slow.cpp) and hence slower C++ compilation (more than 1.5x slower). I didn't find useful information in changelog and not sure what has been changed.

`verilator --cc --top-module DUT --Mdir obj_dir_DUT --assert -O3 --unroll-count 1000000 --unroll-stmts 1000000 --Wno-UNOPTFLAT --Wno-UNSIGNED --Wno-WIDTH DUT.v`

The 4.036 compilation has several ***_Slow.cpp, but I only applied -O3. 4.024 doesn't have those slow files. I tried to exclude those _Slow.cpp to compile the C++ simluator, but it reports undefined symbol error which means there are probably some useful functions in those slow files.

Is it possible to get back to the same outcome as 4.024?

gezalore, 2020-06-13:
Please see the updated BENCHMARKING & OPTIMIZATION section in the manual. In particular you will want to be setting OPT_SLOW appropriately (with no optimization), which should yield faster compilation (than 4.024) with the same simulation performance .  The relevant parts of the Changelog are the ones talking about --output-split and compilation speed.

gezalore, 2020-06-13:
Closing this as answered, please reopen if the above does not help.

closed 2020-06-13 by gezalore

jsn1993, 2020-06-13:
> Please see the updated BENCHMARKING & OPTIMIZATION section in the manual. In particular you will want to be setting OPT_SLOW appropriately (with no optimization), which should yield faster [...]

As mentioned in the manual, `OPT_SLOW applies to slow-path code, which executes rarely, often only once at the beginning or end of simulation.` Does it mean those slow files are still needed for compilation?

gezalore, 2020-06-13:
Yes, all Verilator output is required by the model.

jsn1993, 2020-06-13:
> Yes, all Verilator output is required by the model.

So you mean I should compile those SLOW files with something like gcc -O0 and FAST files with high opt level gcc -O3, and link them together?

gezalore, 2020-06-13:
Yes, that is the idea, though as the manual suggests -O3 will likely yield long compile times with no real performance benefit (use -Os or -O2 instead). The standard verilated.mk has sane defaults and does this for you already (if you are using `make -C obj_dir -f vDUT.mk` or similar).

jsn1993, 2020-06-13:
Thanks. We have our own verilator building mechanism which doesn't use the .mk file directly and that might be the reason why I got very confused when the code structure is changed.

jsn1993, 2020-06-13:
I realized that compiling multiple files is much slower than compiling a single file due to the g++ launch overhead ...

gezalore, 2020-06-13:
I was assuming you are building in parallel. If not, see what verilated.mk does when VM_PARALLEL_BUILDS is not set. Setting --output-split 0 will result in fewer files but at the size where multiple files would be emitted compilation time should dominate, that is unless you have some seriously long launch times on your environment.

jsn1993, 2020-06-13:
After playing around with different options, I found that compiling one monolithic single file is as fast as --build -j4 ...