verilator/verilator #2421: Test suite fails on several archs

aelmahmoudy, 2020-06-13:
When building verilator 4.036, the test suite run fails on the following archs:
armel, armhf, i386 & mipsel

Relevant error seems to be:

{{{
 make[2]: Entering directory '/<<PKGBUILDDIR>>/test_regress'
 make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
 g++ -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/<<PKGBUILDDIR>>/test_regress/../include -I/<<PKGBUILDDIR>>/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
 ar -cr Vtop__ALL.a Vtop__ALL.o
 ranlib Vtop__ALL.a
 In file included from /<<PKGBUILDDIR>>/include/verilated.cpp:25:
 /<<PKGBUILDDIR>>/include/verilated_imp.h: In static member function ‘static IData VerilatedImp::fdNew(const char*, const char*)’:
 /<<PKGBUILDDIR>>/include/verilated_imp.h:490:78: error: no matching function for call to ‘max(long unsigned int, std::vector<_IO_FILE*>::size_type)’
   490 |             const size_t start = std::max(31ul + 1ul + 3ul, s_s.m_fdps.size());
       |                                                                              ^
 In file included from /usr/include/c++/9/deque:60,
                  from /<<PKGBUILDDIR>>/include/verilated_heavy.h:29,
                  from /<<PKGBUILDDIR>>/include/verilated_imp.h:29,
                  from /<<PKGBUILDDIR>>/include/verilated.cpp:25:
 /usr/include/c++/9/bits/stl_algobase.h:222:5: note: candidate: ‘template<class _Tp> constexpr const _Tp& std::max(const _Tp&, const _Tp&)’
   222 |     max(const _Tp& __a, const _Tp& __b)
       |     ^~~
 /usr/include/c++/9/bits/stl_algobase.h:222:5: note:   template argument deduction/substitution failed:
 In file included from /<<PKGBUILDDIR>>/include/verilated.cpp:25:
 /<<PKGBUILDDIR>>/include/verilated_imp.h:490:78: note:   deduced conflicting types for parameter ‘const _Tp’ (‘long unsigned int’ and ‘std::vector<_IO_FILE*>::size_type’ {aka ‘unsigned int’})
   490 |             const size_t start = std::max(31ul + 1ul + 3ul, s_s.m_fdps.size());
       |                                                                              ^
 In file included from /usr/include/c++/9/deque:60,
                  from /<<PKGBUILDDIR>>/include/verilated_heavy.h:29,
                  from /<<PKGBUILDDIR>>/include/verilated_imp.h:29,
                  from /<<PKGBUILDDIR>>/include/verilated.cpp:25:
 /usr/include/c++/9/bits/stl_algobase.h:268:5: note: candidate: ‘template<class _Tp, class _Compare> constexpr const _Tp& std::max(const _Tp&, const _Tp&, _Compare)’
   268 |     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
       |     ^~~
 /usr/include/c++/9/bits/stl_algobase.h:268:5: note:   template argument deduction/substitution failed:
 In file included from /<<PKGBUILDDIR>>/include/verilated.cpp:25:
 /<<PKGBUILDDIR>>/include/verilated_imp.h:490:78: note:   deduced conflicting types for parameter ‘const _Tp’ (‘long unsigned int’ and ‘std::vector<_IO_FILE*>::size_type’ {aka ‘unsigned int’})
   490 |             const size_t start = std::max(31ul + 1ul + 3ul, s_s.m_fdps.size());
       |                                                                              ^
 g++ -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/<<PKGBUILDDIR>>/test_regress/../include -I/<<PKGBUILDDIR>>/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DT_A1_FIRST_CC   -O0 -c -o verilated.o /<<PKGBUILDDIR>>/test_regress/../include/verilated.cpp
 make[3]: *** [/<<PKGBUILDDIR>>/include/verilated.mk:220: verilated.o] Error 1
 make[3]: Leaving directory '/<<PKGBUILDDIR>>/examples/make_hello_c/obj_dir'
 %Error: make -C obj_dir -f Vtop.mk -j exitted with 2
 %Error: Command Failed /<<PKGBUILDDIR>>/bin/verilator_bin -cc --exe --build -j top.v sim_main.cpp
 make[2]: *** [Makefile:38: default] Error 2
 make[2]: Leaving directory '/<<PKGBUILDDIR>>/examples/make_hello_c'
 make[1]: *** [Makefile:252: examples] Error 10
}}}

See: https://buildd.debian.org/status/fetch.php?pkg=verilator&arch=armel&ver=4.036-1&stamp=1592053669&raw=0

stephenry, 2020-06-14:
I'll fix this.

wsnyder, 2020-06-28:
#2435 is fixed so could you please give this another try using the latest git master?

gezalore, 2020-06-28:
Reopen until confirmed

wsnyder, 2020-08-10:
No response, assuming got resolved.

commit 98b5417e (2020-06-28): Disable argument template type deduction. (#2426)

Fixes #2421, #2435 relating to 32-bit build failures.
closed 2020-08-10 by wsnyder