verilator/verilator #2422: How to suppress this warning in 4.036 (never saw this in 4.024)?

jsn1993, 2020-06-13:
How may we help - what is your question?
I have a piece of code that has to dump the vcd twice at one timestamp. With 4.036 I get this warning message. I didn't find the specific -Wno option that only supresses this warning ... 
`%%Warning: previous dump at t=750, requesting t=750, dump call ignored`

Thanks!

gezalore, 2020-06-13:
There is now way to suppress that waring. Every repeated call to dump on a trace file with the same or lower time value than the a previous call is ignored. You should fix your code to call dump with strictly intrementing time values. If this is an issue please explain in more detail about why.

jsn1993, 2020-06-14:
Hmmm. I think it is totally possible to call dump twice at the same time value. The test bench needs to change the input value of the verilated model at some point in the cycle, and an output port combinationally depends on the input port value so I call dut->eval() to propagate the value change. Then I check the output value to see if it's correct. If it's not correct I will terminate the simulation right away, but I want to be able to debug using the updated vcd.

gezalore, 2020-06-14:
I agree you can call eval multiple times within the same time step. The problem is that neither of the supported tracing formats (VCD or FST) explicitly support delta cycles and I am unaware of a viewer that displays anything but the final value in the timestep. Considering that and that dump is generally very expensive, it is best to call dump at the point you actually want to observe, otherwise you will be wasting simulation time. In your example, it sounds like you want to call dump only if you are about to terminate if your combinational outputs are incorrect (so you can debug), or at the end of the time step otherwise. Do you agree?

jsn1993, 2020-06-14:
> I agree you can call eval multiple times within the same time step. The problem is that neither of the supported tracing formats (VCD or FST) explicitly support delta cycles and I am unaware of a [...]

Agree. I guess making it a suppressable warning might be a good thing.

closed 2020-06-14 by jsn1993