verilator/verilator #2424: Fix consts from #2370

wsnyder, 2020-06-14:
@elgorwi , Older Ubuntu is unhappy with the vpi parameter patch, could you please make a pull to fix it, thanks. 
https://github.com/verilator/verilator/compare/19c2906a64464d68f6465db6f2161851c29db078...8006b5479c6716a3c7edf1d1478b5e672c797b6d

elgorwi, 2020-06-14:
Could you provide some more details please (error message, ubuntu version)
Thanks

wsnyder, 2020-06-14:
Sorry pasted wrong link above. See https://travis-ci.com/github/verilator/verilator/builds/171250525 then click on a failing test then can see log.

```
In file included from Vt_trace_param__Syms.h:14:0,
                 from Vt_trace_param.cpp:6,
                 from Vt_trace_param__ALL.cpp:3:
Vt_trace_param_my_module_types.h:29:33: error: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [-Werror]
     const IData var_MY_PARAM2 = 0xcU;
                                 ^
```

elgorwi, 2020-06-14:
Thanks for the details. I’ll look into it.

elgorwi, 2020-06-23:
A fix has been merged, so I think this issue can be closed.

closed 2020-06-24 by gezalore