verilator/verilator #2425: Behaviour of explicit X condition is strange

gezalore, 2020-06-14:
Verilating:
```
 module foo(
   input wire clk,
   output reg o
 );
   always @(posedge clk) begin
     if (1'bx) begin
       o <= 1'd1;
     end else begin
       o <= 1'd0;
     end
   end
 endmodule
```

Yields:
```
 VL_INLINE_OPT void Vfoo::_sequent__TOP__1(Vfoo__Syms* __restrict vlSymsp) {
     VL_DEBUG_IF(VL_DBG_MSGF("+    Vfoo::_sequent__TOP__1\n"); );
     Vfoo* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
     // Body
     vlTOPp->o = 1U;
 }

  void Vfoo::_eval(Vfoo__Syms* __restrict vlSymsp) {
     VL_DEBUG_IF(VL_DBG_MSGF("+    Vfoo::_eval\n"); );
     Vfoo* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
     // Body
     if (((IData)(vlTOPp->clk) & (~ (IData)(vlTOPp->__Vclklast__TOP__clk)))) {
         vlTOPp->_sequent__TOP__1(vlSymsp);
     }
     // Final
     vlTOPp->__Vclklast__TOP__clk = vlTOPp->clk;
 }
```

Which is exactly the opposite of the standard, but more interestingly, --x-assign has no effect on this code. Reading V3Unknwon I was hoping the condition would just be replaced with a random value with --x-assign unique, but that does not seem to be the case. Is this intended?

wsnyder, 2020-06-14:
The intent is x-assign will control this (0/1/random). I'm pretty sure this used to work....

gezalore, 2020-06-14:
So is it fair to then read x-assign as "anywhere you have a literal x value expression, plug in 0/1/random instead" (notably not in casex)? that would be consistent with your statement.

gezalore, 2020-06-14:
Asking because the name/docs suggest that the `--x-assign` option is only about explicit X assignment, but that does not seem to be the intention based on what you just said.

wsnyder, 2020-06-14:
Generally any X should be replaced with a random assignment (e.g. x-assign means make an assignment to Xs, not fix only assignments).

PR #2438 (2020-06-20): Do not fold IF with 'bx condition before V3Unknown
Fixes #2425

Will merge when Travis is happy.
commit aa595d40 (2020-06-20): Do not fold IF with 'bx condition before V3Unknown (#2438)

Fixes #2425
closed 2020-06-20 by gezalore