verilator/verilator #2427: output FIRRTL

simdream, 2020-06-16:
noticing FIRRTL (or maybe more precise, LoFIRRTL) simulator claims even higher run time performance, any way to output FIRRTL from verilator? I tried to use yosys , rading design and output to FIRRTL. But the yosys does not support more SV syntax than verilator.  Maybe the FIRRTL can be output from some middle stage of verilator process before everything is translated to C++?

wallento, 2020-06-16:
May relate to #2084 
But generating it as output is not really in the scope of Verilator. What exactly are you expecting from it? User Verilator as a System Verilog frontend?

wsnyder, 2020-06-16:
I don't believe FIRRTL claims higher performance, and that seems inconsistent with their architecture AFAIK, can you point to data/references otherwise?

simdream, 2020-06-16:
#2084 is for taking FIRRTL as input to verilator. This #2427 is to output FIRRTL from verilator. 

For the performance, I really am not sure it is an apple-to-apple comparison. Just in a previewed paper mentioned a quite impressive FIRRTL simulator speed. I can check if the paper could be disclsoed or not.

wallento, 2020-06-16:
That's why I said "may relate" ;)
Well, let's wait until we have an idea what the actual benefit may be.