verilator/verilator #2429: Generated waivers for multiline are incorrect

wallento, 2020-06-16:
Multiline warnings cause issues with waiver files, for example `VARHIDDEN`
```
%Warning-VARHIDDEN: ../src/openhw_cv32e40p_core_0.1/rtl/riscv_id_stage.sv:57:13: Declaration of signal hides declaration in upper scope: 'WAPUTYPE'
   57 |   parameter WAPUTYPE          =  0,
      |             ^~~~~~~~
                    ../src/openhw_cv32e40p_core_0.1/rtl/include/apu_core_package.sv:38:13: ... Location of original declaration
   38 |   parameter WAPUTYPE         =  0;
      |       
```
Truncate and add a wildcard star solves it, maybe this should be done when writing.

wsnyder, 2021-09-21:
@keithcolbert would be great if you'd like to submit a pull request for this.  I think the suggestion to simply replace everything following a newline with .* would do it.  Plus add the test you already suggested in #3137.  Thanks!

keithcolbert, 2021-09-22:
Ah I missed the dup. PR is linked

closed 2021-09-22 by wsnyder