verilator/verilator #2430: Lint: DECLFILENAME warning is picking up a blackboxed module inside the actual module

imphil, 2020-06-17:
Have a look at this file:

https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim_xilinx/rtl/prim_xilinx_clock_mux2.sv

Simplified:

```systemverilog
module prim_xilinx_clock_mux2 (
  input        clk0_i,
);

  BUFGMUX bufgmux_i (
    .I1 ( clk1_i )
  );
endmodule
```

`BUFGMUX` is a Xilinx-provided module which is effectively a black-box for Verilator.

This file is always compiled, but not used for Verilator (an implementation is selected through a generate if/else block).

When running this through Verilator lint, I get:

```
%Warning-DECLFILENAME: ../../../hw/ip/prim_xilinx/rtl/prim_xilinx_clock_mux2.sv:17:11: Filename 'prim_xilinx_clock_mux2' does not match NOTFOUNDMODULE name: 'BUFGMUX'
   17 |   BUFGMUX bufgmux_i (
      |           ^~~~~~~~~
                       ... Use "/* verilator lint_off DECLFILENAME */" and lint_on around source to disable this message.
```

I'm not sure what happens here, but the message is strange: the filename of the module has nothing to do with the BUFGMUX module, which is instantiated inside it. And the actual module name prim_xilinx_clock_mux2 matches the file name prim_xilinx_clock_mux2.sv.

```
$ verilator --version
Verilator 4.037 devel rev v4.036-60-gfa828ac9b
```
(A recent master.)

wsnyder, 2020-06-17:
Verilator makes internal notfiundmodules for modules like this. Probably that warning doesn't disable with that (when 'VN_IS(nodep, NotFoundModule)').

Perhaps you could attempt a pull request to fix this with a failing test that it fixes?

imphil, 2020-06-17:
I actually wanted to look into a fix for quite a while but didn't find time for it. At least this issue now gives someone else the chance to beat me to it :smile:

wallento, 2020-06-20:
Increasing the pressure by assigning to @imphil ;)